Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #1: Source Code

geekbozu, 2019-02-18:
I see you have a permissive license on the hardware as noted in the documentation, with that I was wondering if the PIC source code be available for this eventually?

Dfinitski, 2019-02-18:
may be in future

Dfinitski, 2019-05-10:
Sources are available now

RichNeese, 2020-02-07:
DO you have a schematic for this board we want to use it on our custom addin board for ubitx . we are looking at the 5x5 as a addin for low power. but we want to put it n a board that mounts under our ubitx pcb

geekbozu, 2020-02-07:
This still has no license applied to it, 
Are you intending on releasing these assets under a an OSH/OSS license? Or leaving them licensed to you?

RichNeese, 2020-02-07:
> <N7DDC-ATU-100-mini-and-extended-boards@noreply.github.com>
> <comment@noreply.github.com>
> Code (#1)

well we  planned on having your call and my call on the board . statung 
the atu is based on your board.

geekbozu, 2020-02-07:
I was askinng dfinitski, We have wanted to make a kit of this for a while but its just not setup in a way where we can legally do so yet. Attribution or not :/

MOUSE1025, 2020-02-07:
Dave,
Do you share the ATU IP with someone?
Christopher

MOUSE1025, 2020-02-07:
Dave,
Do share the ATU intellectual property with someone?
Christopher

RichNeese, 2020-02-07:
someone is selling kits from russian based on your board.
https://www.ebay.com/itm/Automatic-Antenna-Tuner-7x7-DIY-KIT-ATU-100-by-N7DDC-PIC-16F1938-flashed/193171458870

------ Original Message ------
From: "Richard Neese" <r.neese@gmail.com>
To: "Dfinitski/N7DDC-ATU-100-mini-and-extended-boards" 
<reply@reply.github.com>
Sent: 2/7/2020 1:43:32 PM
Subject: Re[2]: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] 
Source Code (#1)

RichNeese, 2020-02-07:
also just selling boards well over priced ad I know that boards would be 
only 1$  a board at pcbway .

https://www.ebay.com/itm/Automatic-Antenna-Tuner-7x7-PCB-ATU-100-mini-by-N7DDC/202706862349?_trkparms=aid%3D1110001%26algo%3DSPLICE.SIM%26ao%3D1%26asc%3D20140131123730%26meid%3D9140cf58bb024dd5b275ab63ba4f6190%26pid%3D100167%26rk%3D4%26rkt%3D15%26mehot%3Dpp%26sd%3D202736655229%26itm%3D202706862349%26pmt%3D1%26noa%3D1%26pg%3D5411&_trksid=p5411.c100167.m2940

------ Original Message ------
From: "Richard Neese" <r.neese@gmail.com>
To: "Dfinitski/N7DDC-ATU-100-mini-and-extended-boards" 
<reply@reply.github.com>
Sent: 2/7/2020 1:43:32 PM
Subject: Re[2]: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] 
Source Code (#1)

Dfinitski, 2020-02-08:
Hello, people.
This my design like other presented here, go with no any license. It is meaning, that all that info, files, pictures and source code are not protected. Excluding things that I am not the author of and some parts of code, where you can see some license description on the top. That is GNU license requirement - what I have took under GNU, I give under GNU.
So you can do what do you want with this design and I am very surprized that fact you did not read the Uer Manual for this device. The first sentences you can see is highlighted in red  :
**Attention ! There are no restrictions and prohibitions on the use of this material for any purpose, as well as all related materials, the author of which is N7DDC.**

MOUSE1025, 2020-02-08:
Dave,
Thank you for the clarification. I understood the original disclaimer in
the manual, but became confused, with the recent inquiries regarding the
Intellectual Property? I truly appreciate your kind, and generous gesture
regarding the the use of your creation(s).
Kind regards,
Christopher

Dfinitski, 2020-02-08:
Hello, Chris.
The Intellectual Property is staying mine anyway. It is mining that everyone who use this design or some part of them should to indicate everywhere my name or callsing as the Autor, autor of idea, designer, etc.
If someone decides to appropriate authorship, he will experience social censure.

MOUSE1025, 2020-02-08:
Dave,
That's a wonderful idea, I understand your goal and hope that you get the
credit that you deserve.
Kind regards,
Christopher

closed 2020-06-01 by Dfinitski