Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #2: ATU 100 for QRP

S54AM, 2019-06-03:
I have ATU100 extended SMD and works fine. What I miss is setup for QRP needs. Sensitivity for working from 1W to max. 20W. Setup min. power in firmware bellow 05 cause unexpected results as is already described in manual. Is there solution adding extra windings on PWR/SWR core or something else?
Additionally I have issue using OLED 091 because it is not turn on always when I turn power up. In my case every tenth time shows up. I have no experiences using other suggested displays.
Thanks in advance.

73, Marko, S54AM

Dfinitski, 2019-06-04:
Hello.
What exact the unespected result you watch ?

I am using 0.91 OLED with 78 address and 03 type and it works perfectly. I
never heard about this problem before.

David.

S54AM, 2019-06-04:
Hello David,

OLED is blind, nothing on it. After several power ups it works normally. I
have to mention that this is strange for me too. I will send short video
tomorrow.
My addresses are the same 78 and 03 and I have two OLEDs and on both the
same indications.

Marko

Dfinitski, 2019-06-04:
Hello, Marco

I could not help you with this, sorry.

You should to find a mistake by yourself.

Good idea to try watch what happened with digital oscilloscope on lines CLK
and DAT after start the power.

But i will try ask other user on russian forum about this.

David.

Dfinitski, 2019-06-04:
Could you send to me in USA one of this OLEDs ?
Is this not to expensive ?

MOUSE1025, 2019-06-04:
Dave,
I enjoyed your exchange with Marco.
CPR

S54AM, 2019-06-04:
Problem with OLEDs is solved. Google knows everything. Two resistors on OLED 0.91 (R3 and R4) were factory swapped (somewere on far east). I swapped resistors ... and Bingo! 
Anyway, important information for those with similar problems as I had. Thanks David for your effort. 
More details on: [https://www.youtube.com/watch?v=__3InV5tdzM](url)

Could I ask about my first question?

Marko

Dfinitski, 2019-06-05:
Hllo, Marko
R3, R4, R9 have the same value 4k7.  R1 = 56k.

Concerning your first question, please explain with more details what exact
problem you watch.

S54AM, 2019-06-05:
Sorry for confusion. I should be more detailed, because there is dozens of different OLEDs 0.91 with different element numbering on back side. On mine, looking on back side, left is located U2 (3v3 reg), then first row in order C9, D1, R2 and R4. Second row C1, C2, C3, C4, R3, C5, C6 and C7. I swapped R4 and R3 and it works. I can send photo of mine. On QRZ.COM is my address, in reply I can attach photo.

My question is about ATU 100 use for QRP activities. Currently minimal operational power is 5W. What is needed to do to drop minimal synchronisation power to 1W?

Dfinitski, 2019-06-05:
Hello, Marko
If your transmitter has a power around 10 - 20 W you can use all this power
for tuning process.
If 5 W or less, set up in cell 0x05 value 01 - that's meaning that minimum
power for start the process will be from 1 W.
But be sure that the simple output power at least by twice more then
indicated in that cell.
It because in the tune process the output power could be dropped and
process could stop.
Protection system of your tranceiver also could drop the output power when
SWR more then 3.

David

S54AM, 2019-06-06:
Thanks David,

As you know, QRP is up to 5W CW and 10W SSB. My RIG is mcHF M0NKA and couple of others small boxes. I am home brewer and my joy is producing this thing into my small lab for my joy. I am amazed with your ATU and your preparedness sharing your intellectual work for free to our HAM community. All respects to you.
I also noticed your ATU project with internal battery and latching relays. This is my big future interest. 
Yes, everything is clear for me about settings (1W). But, I am just thinking to get more voltage with more secondary  wingdings on PWR/SWR core for small power transmitters. For sure firmware changes are needed in this case to get real data on display. My suggestion is, If this is not too complicated for you, that QRP version of your ATU using latch relays could rise interest in QRP community. I will follow your smart products. 
Sorry for delayed answer. The biggest challenge for me is free time especially at summer.
Again, compliments and all respects.

Marko

MOUSE1025, 2019-06-06:
Dave,
Where are you purchasing the 2512 50 ohm resistor? I have tried Mouser and
Digikey, no luck. I thought I would see where you are getting yours before
I ventured out to Newark and the others?
CPR

S54AM, 2019-06-06:
Mouse1025,
Nearest standardised EIA value for resistors is 49.9 ohm (E96). I am using two 100 ohm in paralel to reach exactly 50 ohm.  

Marko

Dfinitski, 2019-06-07:
Yes, 50 Ohm exactly does not exist or very expensive. Use 49.9 as is. Don't
need to add anything.

MOUSE1025, 2019-06-07:
Dave,
I found one:
CPR[image: image.png]

MOUSE1025, 2019-06-07:
What am I going to need if I want to program my ATU? I assume that the
processor can be programmed right on the ATU board. What hardware will I
need to interface between my computer and the ATU. Will I need an IDE
program and compiler?
CPR

Dfinitski, 2019-06-07:
For programm MCU you will need two things:
!)
https://www.ebay.com/itm/PICKit3-Microchip-Programmer-w-USB-cable-wires-Pic-Kit-3/331941067077?epid=668289631&hash=item4d493ab945:g:XGMAAOSwnHZYcAgs:sc:USPSFirstClass!98003!US!-1

2) https://drive.google.com/open?id=12m2qaR-5bgMg4IJChc4-YXJ7Y6bKInuW

MOUSE1025, 2019-06-07:
Dave,
I am winding up the inductors, in advance of the PCB showing up tomorrow.
The T1/T2 calls out a 1:10 ratio, but gives no turns for the wire?
CPR

MOUSE1025, 2019-06-07:
Dave,
Thanks for the info on the programmer. I have one on order. I should of
asked about this sooner.
CPR

Dfinitski, 2019-06-07:
It is 10 turns, 1 to 10, right

MOUSE1025, 2019-06-07:
Dave,
I thought that was the case, I wanted to make sure that I was not mistaken.
Thanks...
CPR

Dfinitski, 2019-06-08:
Hello, Marko.
Expectually for you - fw version 3.0 where you can to indicate any turn ratio of tandem match.
You can use 1 to 5 ratio to increase voltage on the diodes. 
But be noticed that in this case 1/5 of power your TRX will be dissipated on the tandem-match 50 Ohm resistors.

Dfinitski, 2019-06-09:
Hello, Marko.
Sorry for this mistake. Only 0.2 W will be dessipated on 5 W, not 1/5.

S54AM, 2019-06-10:
Hello David,

Thanks for new firmware and suggestion about T1(1:5) for QRP. I will do
this in next days.
You can expect my reply.

Marko

S54AM, 2019-06-10:
Hello David,

Your V.3 firmware is working well from 1W using settings bellow and modified T1 (1:5).

00 | 78 | 03 | 01 | 15 | 13 | 01 | 00 | 01 | 02 | 00 | 07 | 00 | 07 | 00 | 01 | 00
10 | 00 | 50 | 01 | 00 | 02 | 20 | 04 | 50 | 10 | 00 | 22 | 00 | 44 | 00 | FF | FF
20 | 00 | 10 | 00 | 22 | 00 | 47 | 01 | 00 | 02 | 20 | 04 | 70 | 10 | 00 | FF | FF
30 | 00 | 05 | 00 | 01 | 12 | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF

After several tests I have noticed that, if cell 06 is set to 04 (upper power limit for 40W) instead of default 00, then OLED after SWR sometimes become frozen with word TUNE, not showing SWR numerical value. If I leave it as default, then ATU works normally. Maybe I set something wrong.

Marko

Dfinitski, 2019-06-11:
Hello, Marko

Probably manua is not clear enough. When you have set 04, upper power limit
will be 4W in simple mode and 40 W in High Power Messurement mode (cell 30).

ATU is not freesing, it is waiting for when your power will be less then 4
W.

David

S54AM, 2019-06-11:
Hello David,

If I understand correct, my ATU is set in simple mode because cell 30 is 00. If I want limit upper power to 40W because modified T1 (1:5), I should set cell 30 to 01 and cell 06 to 04.

Dfinitski, 2019-06-11:
Marko, sell 30 is for possibility to measure up to 9999 W and with bigger
step of course. For QRP it is 00 value. In 3.0 fw version you can set
tandem rate freely independently to 30 sell value.

Dfinitski, 2019-06-11:
Marko, probably you read old version manual. Please redownload fresh one.
Also in High Power Messurent mode low limit power is setting as tens watts
too. 01 value is meaning 10 Watts.

S54AM, 2019-06-11:
Thanks David. For QRP operation: simple mode (30=00), tandem match ratio 1:5 (31=05), Min. start up power 1W (05=01), upper power limit 40W (06=40). 
David, could you confirm?

Dfinitski, 2019-06-12:
Yes, it is correct.

closed 2019-06-14 by S54AM

efkleas, 2020-05-24:
Good afternoon from Greece. I need your help. I am facing this problem. After the tuning I have the power amount but not SWR. Just beside the SWR indicator remains the word "tune" as well as L: =0 and C:= 0. The mini tuner seems to work (by pressing the PTT you can hear the relays working). What is wrong? Thanks in advance.

Dfinitski, 2020-05-25:
Hello. After long-click pushing the button "Tune" the tuner is starting a
tune proces. After that you have to provide properly RF power, not below
then Pmin and not above Pmax, wich were determined in MCU cells. The
minimum level by default is 5 Watts. The properly power level should be
present all the time needed to end the tune process.
Other way you can  do short-click the "Tune" button to reset device and
return it to waiting mode.

efkleas, 2020-05-25:
Thanks, David. I did exactly what you said to me. After tuning I did the short-click the "Tune" button to reset the device and return it to the waiting mode. Ok so far but in this mode, I have not SWR or power numbers when PTT.  I don't know what I am doing wrong. My eeprom data as follow.

00 | 7E | 01 | 01 | 15 | 13 | 05 | 99 | 01 | 02 | 00 | 10 | FF | FF | FF | FF | FF
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
10 | 01 | 10 | 02 | 20 | 04 | 50 | 10 | 00 | 22 | 00 | FF | FF | FF | FF | FF | FF
20 | 00 | 10 | 00 | 22 | 00 | 47 | 01 | 00 | 02 | 20 | FF | FF | FF | FF | FF | FF
30 | 00 | 10 | 12 | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF | FF

(We are talking about 100 watts ATU. The board for display 1602 made on the PCF8574AT chip has the address 7E, 1602 LCD with port extension card, device without additional buttons, minimum power cell required to start the setup 5 watts, maximum power for safe tuning 99 watts, tandem-match with a ratio of 1:10 turns)
Please look at my above eeprom data and If necessary make the appropriate corrections. Thanks

Dfinitski, 2020-05-26:
Please make shure you are using the latest version of FW.
"TUNE" is on display after reset or numbers (zero) is on ?
Can you get a power  metering after power-on device and can not after
Tune-Reset process?

efkleas, 2020-05-26:
David in order to give you a whole picture  I uploaded a video of my problem. Please give a look at the video. 
https://www.youtube.com/watch?v=z7ZXkcGE0dU

Thanks 
many 73
SV1CRN

Dfinitski, 2020-05-26:
Thank you.  Now it is much more clear.
You have the mini-board,  not extended.
The tuner in auto mode.  Please check with multimeter pins 26 and 25. If
they have 5 Volts,  do a short to ground pin 25.
The point between two top lines on the LCD should be disappeared.

MOUSE1025, 2020-05-26:
Great job Dave.
Kind regards,
Christopher

efkleas, 2020-05-27:
Thanks, David,
I did exactly what you suggest to me and now I have an ATU that works perfectly. You are the best.
Kind regards
Nikolaos

k2db, 2020-06-01:
I just ordered FULL KIT DIY Automatic Antenna Tuner 7x7 (ATU-100 by N7DDC), what files do i need to construct the kit, do i need anything extra like a display, etc, and where do i go to get the info?
Paul K2DB

n6ln, 2021-01-09:
I need to program the ATU-100 to tune at power levels below 5 watts. I now have the Pickit 3 but I have no instructions on how to use it for this purpose. Can anyone help?
Thanks.  N6LN

clsguy, 2021-01-19:
@n6ln   I am also looking for documentation on how to use the Pickit3.  Did you have some documentation you could share?  Thanks.