Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #4: Programming with Pickit2

victorkoren, 2019-12-24:
I am trying to program the ATU-100 with Pickit-2 but it shows "no device detected".
I set the device manually and it detects that the device has external Vdd, but does not detect the PIC.
Is there a problem using Pickit-2?
Victor - 4Z4ME

Dfinitski, 2019-12-24:
Hello, Victor.
As I know, Pickit 2 does not support PIC16F1938. You have to swap .dll file from Pickit 3 programm or change your programmer to Pickkit 3. After that you should select manually Device Family -> Midrange -> 1.8V min
Because this family uses other Vpp voltage then others.

victorkoren, 2019-12-24:
Hi,
Thanks for the prompt answer, however that is strange because in the Pickit 2 app I can choose the PIC 16F1938 manually, so the option to choose it is there but it is not supported? 
Victor - 4Z4ME

Dfinitski, 2019-12-24:
I am sorry, 1938 is supported, proof
http://ww1.microchip.com/downloads/en/devicedoc/pickit%202%20readme%20v2-61-00%20(a).txt
Please give more info what is happening when you are trying to programm it.

victorkoren, 2019-12-24:
I found a readme DOC from microchip that states that 16F1938 is supported:
http://ww1.microchip.com/downloads/en/devicedoc/pickit%202%20readme%20v2-61-00%20(a).txt
So why doesn't it work?
I saw that other people had this problem.
Victor - 4Z4ME

victorkoren, 2019-12-24:
We posted together the same file...
Anyway I saw that also other people had this problem and didn't solve it.
Tomorrow I will post a screen picture.
Victor = 4Z4ME

victorkoren, 2019-12-24:
It seems that there is a newer DAT file that updates the devices that PICKIT 2 supports. I'll check that tomorrow.

Dfinitski, 2019-12-24:
Looks like that. Let's go !

victorkoren, 2019-12-24:
I updated the data file and it worked ! 
Just wanted to configure the ATU to work in automatic state.
Thanks for help.
Victor - 4Z4ME

closed 2019-12-24 by victorkoren

Dumper88, 2020-11-29:
Hi is it possible to have a link to download PK2DeviceFile the most updated? thanks

Dfinitski, 2020-12-01:
Hi. You can find it here https://drive.google.com/file/d/18wiH5RjL5R8ku7a9MQ4Hxx9SmUZInpvj/view?usp=sharing