Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #5: T1/T2

IAceVenturaIcz, 2020-04-24:
Hello,

may I know how many turns are on T1/T2 please ?
Thank you

2M0LWS, 2020-04-25:
I have done 5 turns on each side. Suitable for 5w minimum power.

Dfinitski, 2020-04-25:
*T1, T2 - TANDEM MATCH, BN43-202 AMIDON CORE* *1:10*
In latest firmware for extended board this ratio might be what you want,
from 5 to 30 typically dependet to desired maximum power.
The limiting criterion is the maximum voltage on analog inputs PIC.
For 1939 it can not to excide 4.096  Volts, for 2520 5Volt with maximum
input power.
For standart 100W it is 1/10 Turns, for QRP below 50W 1/5 Turns preferred.
Voltage on ADC = SQR(2) * Vrf_in / Ratio / 3

IAceVenturaIcz, 2020-04-27:
thank you all for replay. Not sure about how that works. I need 1/10 as been mention, as I will use 100W . What I am nor sure how to wired this toroid. I bit confused as I need to get 4 wires out. Is there any picture how to do it please ?

Dfinitski, 2020-04-27:
The picture on the schematic. Try "tandem match" in the Google Search for
more info.

IAceVenturaIcz, 2020-04-27:
Thank you David, sorted for now...

closed 2020-04-27 by IAceVenturaIcz