Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #7: Automatic Antenna Tuner 7x7 (ATU-100 by N7DDC) Query

k2db, 2020-06-01:
I am a newbie here, looks like fun ,I love kit building, doing it since early 60's,
I just ordered FULL KIT DIY Automatic Antenna Tuner 7x7 (ATU-100 by N7DDC), what files do i need to construct the kit, do i need anything extra like a display, etc, and where do i go to get the info?
Paul K2DB

Dfinitski, 2020-06-01:
Hello, Paul.
This project is fully opened so you can find all the files here on GitHub.
I recommend you to read User Manual firstly.
The schematic would be helpful also.
If you do assembly with no errors, this device will work immediately with
any questions .

David, N7DDC

k2db, 2020-06-01:
thank you, what files do i download ?

Dfinitski, 2020-06-01:
Hi, Paul. Everything you need you can find here
 https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/tree/master/ATU_100_EXT_board

k2db, 2020-06-16:
David, I don't have Microsoft Word, so I can't open the instructions. Is
there any way I can ask you to send me all the info in PDF format?
Building instructions, operation manual, etc.
Still waiting for the kit to be delivered, thought I might get a jump on
what is coming for me, and anything I will need that is extra.
Thanks,
Paul K2DB

Dfinitski, 2020-06-16:
Hello, Paul.
The manual in pdf is present here. Did you try the link in my previous message ? 
Everything you need you can find on the schematic and BOM list.
To open odt files use a free software like Open Office.

Celta1960, 2021-02-18:
Hola amigos, tengo un problema con el atu-100 ver 3.0 el cual funcionar funciona perfectamente pero una en ciertas frecuencias es imposible ponerlo ya que me marca una señal fuerte hasta el punto de no poderlo utilizar, esto ocurre en todas las bandas. 
       Adjunto dos fotos con el puesto y quitado, A que es debido?.
   Un saludo
[image]
[image]

Dfinitski, 2021-02-18:
Hello, Celta1960

**New in the version 3.2
Fixed bug with continuous data transfer to the display.**

So use 3.2 version of firmware.

To check this out, disconnect the display and try without it.

Dfinitski, 2022-08-08:
Hi.
The most likely it uses complicated data exchange via CAT

Dfinitski, 2022-08-08:
But you can try to find native Start and Key signals on your rig.

Plasmastorm73, 2022-08-14:
Are there any step by step build instructions for this kit?

Dfinitski, 2022-08-15:
Hi.
Yes.
1) install all components to the board
2) power on it
3) use it

Plasmastorm73, 2022-08-15:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/7/1214477799@

You know. People like you are causing beginners to give up with your
rudeness. You could have simply said you were too lazy to make a step by
step guide but instead you have to be a jerk.

paul-jewell, 2022-08-15:
> You know. People like you are causing beginners to give up with your rudeness. You could have simply said you were too lazy to make a step by step guide but instead you have to be a jerk.

David has put an inordinate amount of time and effort to make his designs effective and available. As far as I am aware, he doesn't make any money out of the sale of the different kits available from Aliexpress, Banggood, eBay etc, and no doubt has to devote a significant amount of his personal time and effort to supporting people who ask him questions which could already be answered with a bit of googling. He doesn't owe anyone anything - we all have a debt of gratitude to him.

It is a straight forward assembly, and within the skills of anyone familiar with building kits. The tuners are also available fully built for those who don't wish to wield a soldering iron. 

I don't think David was being rude - he was simply being to the point. If you don't have the confidence to build a kit, then buy a complete item - they are still amazing value compared with the competition.

nanko55, 2022-08-15:
Agree.

I’ve build at least three of them, without problems, they all work very well!
For the beginners: take a look at github, Google, AliExpress for tips and schematics, thats how real radio amateurs do it 😉
If you don’t like or want to do this, purchase an ATU from ICOM or MFG.
I think David did an outstanding job👍

Regards, Nanko van Dijk, PA0NVY

Plasmastorm73, 2022-08-15:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/7/1214717781@

Dear Paul. This wasn't for me. I asked a simple question and he was snotty
about the answer. PERIOD! A simple no there isn't would have been straight
forward. I was looking into this kit for a 12 year old kid I am teaching.
Maybe you should mind your own business and leave the conversation to the
adults. I've been building Oscilloscopes for a MAJOR brand for over 30
years and also teach soldering and electronics building classes. Some
people who just assume the facts like you are the real problem.

paul-jewell, 2022-08-15:
Hi William,

I understand it wasn't for you - I could see from your QRZ page that you 
have many years of electronics experience and don't need instructions.

In my opinion, Dave was succinct and to the point. What else could he 
have said?

If you look at the other comments above yours, you will see that even 
when he takes to respond, people don't always pay attention to what he 
says. /Comments like yours/ will ultimately drive creative people like 
Dave away - he doesn't have to share anything, or do anything for anyone 
else. *He owes us nothing, and we should be eternally grateful for his 
generosity!*

I have seen a number of people who make the product of their hard work 
available for free to others, and then get lambasted for it. In some 
cases, this has led to them withdrawing their information and support, 
to the detriment of the community.

If you think there should be instructions, then create some and make a 
pull request so they can be added to the github repository. Dave doesn't 
need instructions for himself, so why should he write any?

I have nothing more to say about this, as I doubt I can change your 
views, and it's not constructive.

Best regards,
Paul

Plasmastorm73, 2022-08-15:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/7/1214953418@

You completely miss my point. I don't give a damn if there are or aren't
any instructions. MY point was the asshole way he said no there weren't
any. "Dave was succinct and to the point" was total bullshit. He was being
a snarky asshole instead of just saying no there aren't any. I neither
found it amusing nor helpful.

Anyway. You are correct that you won't change my mind on this. I didn't get
my panties in a twist over there being no instructions like you are
implying. I got angry at his attitude and asshole reply.

MOUSE1025, 2022-08-15:
> ID: <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/7/1215009197@

No good deed goes unpunished!
Great work Dave.
Kind regards,
Christopher

davidhay45, 2022-08-15:
@Plasmastorm73 you have demonstrated to all reading this thread that YOU are the jerk around here.  David N7DDC has done a wonderful job in designing these ATU's and release all of the information to the world at large and does not deserve criticism from pathetic idiots like you.

Dfinitski, 2022-08-15:
That's a good lesson for me. Stupid questions SHOULD be ignored.

Plasmastorm73, 2022-10-11:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/7/1214740123@

I have no issue with the kit, nor the amount of time he put into it. I have
issue with people giving snotty answers when it isn't called for. PERIOD!