Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #8: Tandem match T1   T2

morseoneuk, 2020-06-09:
Is wire gauge critical?  10:1   ok,   but what wire diameter?

Dfinitski, 2020-06-09:
AWG 32 in PTFE silver core works perfect.

morseoneuk, 2020-06-09:
Hi David, many thanks for the info..  that is  just what I needed to know.best regards   Scotty

fedemartini, 2020-07-15:
Sorry for the dumb question, but I wasn’t able to find anywhere the data on T1/T2. 1:10 ok, but how many turns? 
Many thanks

Dfinitski, 2020-07-15:
Hello. Ask Google "tandem match" and you will find everything you need,  with pictures.
How many turns for 1:10 tandem ? 1 and 10 turns of course.

stefan75dc4, 2020-08-26:
Hello David,
i buyed the Tuner on ebay, but the 100 Watt - Version (7x7 Elements).
Can I make a 1000 W Tuner with this PCB or I need another PCB-Board.
How much Turns i need for the tandem match? 1:10 or a other worth?
Sorry for this question, but this is my first greater Kit for Amateurradio.
I must collect further experience.
Thanks in forward.
Vy 73 de DC4KJS (Stefan)

Dfinitski, 2020-08-27:
Hello, Stefan
You can find info about 1kw tuner here, use google translate . Boards are present on the Ebay. http://www.cqham.ru/forum/showthread.php?38456-ATU-N7DDC-%E2%E5%F0%F1%E8%FF-1Kw-%EE%F2-EU2AV

Dfinitski, 2020-09-14:
Stefan, 1kW tuner here
https://www.youtube.com/watch?v=RaizIevy-BI

stefan75dc4, 2020-09-14:
Hello David,

Thank You for the Link.

Vy 73 de and best wishes 

Stefan Dc4kjs 

Holen Sie sich Outlook für Android

https://www.youtube.com/watch?v=RaizIevy-BI