Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #9: Blown capacitor C27 or C26

MartinDavidWaller, 2020-06-21:
Hi,

I've recently built one of the kits and it's all been going well until today. I am driving it from a KX3 with 10w output and found that the tuning could hang if the power went below 5, possibly due to the KX3 reducing power out to protect itself. To resolve this I've change the EEPROM address 5 from 5 to 1 and it seemed to be much happier. 

I was using it to tune an antenna on 24Mhz and was half way through a CW QSO when the fuse on my power supplt blew and the shack filled with 'an interetsing smell'! On investigation then only damage seems to be C27 or C26 that has blown. The ATU still works, I can enter test mode and cycle through the L and C values.

Any advice or replacing the cap would be appreciated, do I need a better rated cap? 

Many thanks for any advice you can give.

Martin
G0PJO

Dfinitski, 2020-06-21:
Hello, Martin.
Thank you for sharing that.
Yeah, this is very wierd to see blown DC current capacitors (???!!!).
I don't see any other reason to do that except all of them were very low quality or were rated for low voltage, 5-6 Volts.
Just change them to 25-50 Volts rated .

MartinDavidWaller, 2020-06-22:
Hi David,

Many thanks for the reply. Yes, very odd, I'll order some and replace them. Many thanks for the help.

Martin

closed 2020-08-27 by Dfinitski

MartinDavidWaller, 2020-08-27:
Hi David,

Sorry for not getting back to you on this. I replaced the capacitor and it has worked like a dream ever since! Many thanks.

Martin