Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #10: Manual control

pa3hea, 2020-06-29:
Hello,
I build the tuner and it works reasonably well.
It happens on a regular base that he mis the lowest swr.
Is there a posibility to control the capacitors and the coils manualy to fine tune the swr.
Grts
Pa3hea

Dfinitski, 2020-06-30:
Hello.
You can use the manual mode , (Bypass + Auto + Power On), to find the best LC combination but there is no this future in working mode.
Try to encrease Relay delay time, it would be helpful.

pa3hea, 2020-06-30:
Tnx for your answer, can you tell me how to change that.
Do I need to reprogram the pic for that.

Tnx

Bart (pa3hea)

Dfinitski, 2020-07-01:
Yes, read the manual, cell's description.
But I have better idea. Add 1-3 meters of cable to your RF antenna fider  (after ATU)  and try again.

pa3hea, 2020-07-01:
Hello David,Tnx for your attention.I understand what you mean with that.Its not realy a big issue when make a mistake because it is simple to correct when it happens.I am very satisfied with it.It does his job verry well.Well done david my compliments.If you dont mind I want to write a little peace about in a technical digital magazine here in holland called "razzies" about my findings and building experience.Most of the radio amateurs of the local clubstation PI4RAZ write peaces about several kinds of projects they develop and build.Every month there is a edition that we spread all over the world in 2 languagesThe author of this magazine is Frank PA3CNO.Can you let me know if you find this a problem or not.TnxBart (PA3HEA)Verzonden vanaf mijn Samsung Galaxy-smartphone.<div>
</div><div>
</div><div>-------- Oorspronkelijk bericht --------</div><div>Van: David Fainitski <notifications@github.com> </div><div>Datum: 01-07-2020  22:15  (GMT+01:00) </div><div>Aan: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards <N7DDC-ATU-100-mini-and-extended-boards@noreply.github.com> </div><div>Cc: pa3hea <bart@bwautomatisering.nl>, Author <author@noreply.github.com> </div><div>Onderwerp: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10) </div><div>
</div>Yes, read the manual, cell's description.
But I have better idea. Add 1-3 meters of cable to your RF antenna fider  (after ATU)  and try again.

Dfinitski, 2020-07-01:
Hello, Bart.
There are no any problem to use this project information for any purpose.

pa3hea, 2020-07-01:
I have a stuppid brainfart, would it be  possible to convert the software for the pic controller to arduino and replace the pic by a arduino.Because arduino is verry populare ad the moment here.Dont understand me wrong its just a brainfart and I dont know how to do it.Just curious if this is possible.TnxBart (PA3HEA)Verzonden vanaf mijn Samsung Galaxy-smartphone.<div>
</div><div>
</div><div>-------- Oorspronkelijk bericht --------</div><div>Van: David Fainitski <notifications@github.com> </div><div>Datum: 01-07-2020  22:52  (GMT+01:00) </div><div>Aan: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards <N7DDC-ATU-100-mini-and-extended-boards@noreply.github.com> </div><div>Cc: pa3hea <bart@bwautomatisering.nl>, Author <author@noreply.github.com> </div><div>Onderwerp: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10) </div><div>
</div>Hello, Bart.
There are no any problem to use this project information for any purpose.

Dfinitski, 2020-07-01:
PIC controllers are very popular in the world and have low price. You should to be a crazy Arduino's fan to get hours brainstorm to convert the code. But I think it is not a big problem.

DIETMAR4, 2020-08-03:
Hello !
I built the ATU 100 7x7. It works very well. Now I have a question. If I want work with my power amplifier (1 kw PEP) I shurely have to switch off the ATU 100. But how can I do this without removing of the PL259 Plugs ? Thanks for your answer !
Dietmar4 (DK8TT)

Dfinitski, 2020-08-03:
Hello.
Use Bypass mode or power off the tuner board.

DIETMAR4, 2020-08-03:
Hello dear David,
thank you very much for your quick answer. May I ask back, if I switch off the power of the tuner, the power of the PA (1kW) really cannot destroy the tuner ?  Then I have a second question to Bypass: If I make a contact (switch) from RP1 to ground. Which function has this step ? 
Thank you very much for your answer.
Vy 73
Dietmar (DK8TT)

Von: David Fainitski
Gesendet: Montag, 3. August 2020 22:14
An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards
Cc: DIETMAR4; Comment
Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello.
Use Bypass mode or power off the tuner board.

pa3hea, 2020-08-03:
Maybe stupid question but why do you want to put 1kw through the small 100 watt tuner.Verzonden vanaf mijn Samsung Galaxy-smartphone.<div>
</div><div>
</div><div>-------- Oorspronkelijk bericht --------</div><div>Van: DIETMAR4 <notifications@github.com> </div><div>Datum: 03-08-2020  22:55  (GMT+01:00) </div><div>Aan: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards <N7DDC-ATU-100-mini-and-extended-boards@noreply.github.com> </div><div>Cc: pa3hea <bart@bwautomatisering.nl>, Author <author@noreply.github.com> </div><div>Onderwerp: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10) </div><div>
</div>Hello dear David,

thank you very much for your quick answer. May I ask back, if I switch off the power of the tuner, the power of the PA (1kW) really cannot destroy the tuner ?  Then I have a second question to Bypass: If I make a contact (switch) from RP1 to ground. Which function has this step ? 

Thank you very much for your answer.

Vy 73

Dietmar (DK8TT)

Von: David Fainitski

Gesendet: Montag, 3. August 2020 22:14

An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards

Cc: DIETMAR4; Comment

Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello.

Use Bypass mode or power off the tuner board.

—

https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/10#issuecomment-668236511

BW automatiseringWillem Dreeslaan 2452729 NE ZoetermeerT) +31 798 200 240F) +31 78 200 241info@bwautomatisering.nlwww.bwautomatisering.nl
Op al onze producten en diensten gelden onze algemene voorwaarden voor B2B en consument zoals vermeld op de website.

Dfinitski, 2020-08-03:
Hello.
No, you can not to put 1kW thought the tuner board.
The tuner board should be connected between your 100W rig and 1kW PA. An antenna is connected to PA output. When you don't use a PA, it is bypassed (usually, should be). When you use a PA - Bypass the tuner board.

Dfinitski, 2020-08-03:
There are two additional buttons - Auto and  Bypass.

DIETMAR4, 2020-08-04:
Dear David,
thank you very much for your precious answer. Now I changed the position of the tuner between the rig and the PA. - I don’t have the button Auto und Bypass. I only use the button Tune. Can you tell me please, how I can install the Buttons Auto and Bypass. Have I to connect a switch between point RB1 and ground and Point RB2 and Ground ? And what are the functions of these two buttons ? Thank you very much for your patience to my stupid questions.
Vy73
Dietmar (DK8TT)

Von: David Fainitski
Gesendet: Montag, 3. August 2020 23:11
An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards
Cc: DIETMAR4; Comment
Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello.
No, you can not to put 1kW thought the tuner board.
The tuner board should be connected between your 100W rig and 1kW PA. An antenna is connected to PA output. When you don't use a PA, it is bypassed (usually, should be). When you use a PA - Bypass the tuner board.

Dfinitski, 2020-08-04:
Hello.
You can use a reset tuner instead Bypass. A Bypass mode allows you to return to LC settings anytime, unlike a reset.
Yes, just connect the additional buttons to ports and ground. Install 0.1uF caps over there also.
You can read about additional buttons in the user manual. You can see how to connect additional buttons on the schematic.

DIETMAR4, 2020-08-04:
Dear David,
thank you very much for explaining of the two buttons and its installing. Today I talked to an OM from southern Germany. He also wants to build your ATU-100.
Now I thank you very much for your kindness help. I wish you all best and much health !
Vy 73 
Dietmar (DK8TT)

Von: David Fainitski
Gesendet: Dienstag, 4. August 2020 17:05
An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards
Cc: DIETMAR4; Comment
Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello.
You can use a reset tuner instead Bypass. A Bypass mode allows you to return to LC settings anytime, unlike a reset.
Yes, just connect the additional buttons to ports and ground. Install 0.1uF caps over there also.
You can read about additional buttons in the user manual. You can see how to connect additional buttons on the schematic.

DIETMAR4, 2020-08-15:
Dear David,
I have another question. I see, that the swr-value of ATU-100 often is different from the value of my swr-meter. It is connected behind the atu-100. How can it be ?
Thank you very much for your answer.
Many 73
Dietmar

Von: David Fainitski
Gesendet: Dienstag, 4. August 2020 17:05
An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards
Cc: DIETMAR4; Comment
Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello.
You can use a reset tuner instead Bypass. A Bypass mode allows you to return to LC settings anytime, unlike a reset.
Yes, just connect the additional buttons to ports and ground. Install 0.1uF caps over there also.
You can read about additional buttons in the user manual. You can see how to connect additional buttons on the schematic.

Dfinitski, 2020-08-15:
Hello, Dietmar
The lines in the tuner board have a parasitic inductance, relays -
parasitic capacitance, so nothing weird if you see different SWR values in
that heterogeneous transmission line.
Connect your SWR meter before the tuner as close as it's possible to see
more valuable values.

DIETMAR4, 2020-08-15:
Dear David,
thank you very much for your excellent advice. Now I changed the position of the SWR-Meter and can see, that the different SWR-values are much closer !
I wish you a beautiful Sunday tomorrow und much health.
My best 73 to you
Dietmar

Von: David Fainitski
Gesendet: Samstag, 15. August 2020 17:41
An: Dfinitski/N7DDC-ATU-100-mini-and-extended-boards
Cc: DIETMAR4; Comment
Betreff: Re: [Dfinitski/N7DDC-ATU-100-mini-and-extended-boards] Manual control (#10)

Hello, Dietmar
The lines in the tuner board have a parasitic inductance, relays -
parasitic capacitance, so nothing weird if you see different SWR values in
that heterogeneous transmission line.
Connect your SWR meter before the tuner as close as it's possible to see
more valuable values.

closed 2020-08-27 by Dfinitski