Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #11: Under rated capacitors C7 1n, C6 470pf, C7 220pf

pa3hea, 2020-07-22:
Hello david,

I notished that c5 c6 and c7 are under rated.
On the lower bands like 80m band
They get verry hot in a view seconds when you use 100 watt. And the swr is moving when that happens.
I replaced them with high Q low ESR  500v types from johanson.
The capacitor kit P/N from Johansons is S111DUE.
This capacitors solve the problem.
The capacitors witch are included in the kit are most likely 50v capacitors i ques if I look to the size of the capacitors.
If everybody get this capacitors then you get a lot of complains I think.

Grts

PA3HEA

Dfinitski, 2020-07-23:
Hello.
ATC800 is the best solution in this case, but expensive.
If you use simple 1206 capacitors, choose them with 1kV and higher rating with white or rose ceramic. Don't use brown or grey ones.
C5, C6, C7  must be composed with lower values caps, not more then 220p, because bigger values used bad dielectric usually.
NP0 dielectric must be used in all C1 - C7 capacitors.

pa3hea, 2020-07-23:
I replaced them with the same type you advice 2828.
That works fine.
But you advice also not to use brown capacitors, all capacitors c1-c7 are brown in the kit and 1206 so i'm not sure if that is what you advice in this.

Grts

Bart (Pa3hea)

Dfinitski, 2020-07-23:
Hello.
I am designer of this project, I am not a seller of kits . All my
suggestion you can read in my previous post.
Please ask seller of your kit if you have some problems with that .

closed 2020-08-27 by Dfinitski