Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #12: Compile the project with a free IDE?

khkremer, 2020-08-10:
I don't have any experience with development for PIC processors, so please excuse my ignorance regarding development environments. I have successfully used the MPLAB X IPE to program the hex file into the chip, but now I'd like to play around with the source code. I tried to load the project into MPLAB X IDE, but it does not recognize the project folder as a valid project. That prompted me to poke around in the project configuration files, and it looks like you are using MicroC, which is a bit on the expensive side for just playing with one project. Do you have any suggestions for getting this to compile using a free system? 

Thanks, 
Karl Heinz - K5KHK

Dfinitski, 2020-08-10:
Hello, Karl
Of course you can create the new project in MPLAB IDE, copy all .c and .h files to it and try to compile.
You will need to change native MicroC libraries like ADC conversion, Types Conversion , EEPROM and probably something else, I don't remember.
I2C is not a library, it is software realization.
Use a free demo version of MicroC to obtain info about PIC configuration, library description, etc.

sparkey2450, 2020-11-28:
You guys are kind of throwing me here. You mentioned MicroC but I found MikroC. Is the second one correct? I like what I see with this tuner. What I'd like to do is set it up so I can access the parameters from the shack via WiFi with the tuner Mtd at the antenna. Been a while since I did any programming so this will be a big learning curve.

Dfinitski, 2020-12-01:
Hi, Steve.
Yes, you are right, MikroC is correct name.
It would be interesting to do something on ESP32 Arduino board.

rogerclarkmelbourne, 2020-12-01:
The EPS32 is a great device, but be aware that when the WiFi it enabled it takes loads of current. Hundreds of mA, like all Wifi devices do.

You'd also need to check the accuracy of the ADC on the ESP32. It may be OK, but it possibly may be worse than the PIC.

I'm also not sure how hardened the Wifi Rx is on the ESP32 and how well it would react to 100W passing by in close proximity, and conceivably it could actually damage the Rx.

sparkey2450, 2020-12-01:
Ok thanks for the info. You can email me direct if you look up my QRZ page for W8TTY I've got a few ideas. Been a little bit of a trip down memory lane for me. I haven't done any real programming since I retired back in 2004. My current project is a Mag Loop. Another subject. I have some ideas and for real quick I see two boards.  Your design mated to an Arduino or Raspberry Pi. Most likely going to take some case design and shielding. All the commercial Auto Tuners are designed to sit on a desk. The other ones that are at the antenna lack any kind of real interface. I have both compilers loaded and got sticker shock when I looked at the price of the optimizing compiler from MPLAB. The one you are using is reasonable for hobby use. The Free MPLAB Compiler when I loaded the project and ran it had lots of ????? Warning Warning Will Robinson. That is most likely Operator error. I got the other one to run just fine but since its a demo it will not build the executable because of size. Anyway this will be a fun project. Still getting this old Unix, Microsoft C Programmer back on his feet. Thanks for a great job on the design. I'm not one to re-invent wheels so the reason I was looking at the Raspberry PI is all that VNC and WIFI stuff is there and waiting to be used.  I have a ATU-100 box coming this Friday. Supposedly wired by you. Have a great day. As for powering this thing I'll be looking at Solar and batteries.  Nice job again on the design.

Dfinitski, 2020-12-01:
Steve, I don't want to do something else on this project. This tuner was created as a "black box" under antenna and I have add the display only for debugging data output. but people asked me add something else more and more, so as result you can see what we have. On my personal opinion, this tuner should be "blind", next to antenna in full automatic mode.
All the info what I need I can see on my transciever's display.
It would be interesting for me how far the human mind can go in perfecting a simple thing.

sparkey2450, 2020-12-01:
> Steve, I don't want to do something else on this project. This tuner was created as a "black box" under antenna and I have add the display only for debugging data output. but people asked me add [...]

Ok, Thanks for a great piece of equipment. I appreciate the work you have done. It is very well designed. I take a little different approach. I like to know exactly what going on at the end of the coax. Thanks again for all your hard work. 

Steve W8TTY

sparkey2450, 2020-12-04:
David,  are you willing to answer a few questions about the code as I read through it? If not then that's fine. I'm like anyone else. I can get lazy and look for the shortest path. So far so good on the code. I'm looking at Firmware 3.1. Unless you suggest one of the others. Thanks
Steve

Dfinitski, 2020-12-04:
Hi Steve.
I am open  for questions.

sparkey2450, 2020-12-04:
Thanks David. I got the box today and tried it out from the shack worked well. It only has the Tune Button. It tunes quickly. Yes, I've already had the top off. Version 3.0 Firmware is installed. I was assuming I could add the Auto Button to the front but not sure on the board where that connects at this point. But I noticed the OLED and Program header being used on the board is bent over and wires are soldered directly to the header. That was a surprise. I'll figure out how to get the PICkit3 connected. I want to put it in "Auto" Then put it out at the antenna. I only have one software question at his point. There are some Forever While Loops "while(1) { bla bla bla }; in the code that appears to be part of the testing stuff. I did not see anything obvious to me on a break out of the loop.  Is that just for the testing and once entered the only way out is power down or reset? Or am I missing something? 

I have looked at the MPLAB Compiler and I think I will stick with mikroC. 

I won't bother you any more this December. I'm sure you got things going on and so do I . I've got the Mag Loop stuff I have not finished. Then in Jan. I'll get back to this . My goal is to have this at the antenna and have it take care of matching automatically.  Whether I ever get to the point where I can see the data in the shack is another project. I appreciate your help.

Thanks,
Steve W8TTY

Dfinitski, 2020-12-04:
Hi, Steve.
Those are infinite loops for Fast test mode, manual mode and fider loss
setting mode. (In the 3.1 firmware).

sparkey2450, 2021-01-20:
David,
Happy new year hope all is well for you. I saw something in sharp_cap(). Int mem_swr; but down in the code its mem_SWR; Am I missing something? I've been porting this to the MpLab Compiler XC8. Got it to compile today. Have not tested it. I stripped out the things I do not need. If you want a copy send an email directly to my email address. My email address on my QRZ is correct. Thansk
Steve W8TTY

Dfinitski, 2021-01-20:
Hello, Steve.
Did you mean *min_swr* ? My compiler allows to use these names, it is not a
mistake for MikroPic. Actually *int min_SWR* , you are right.

sparkey2450, 2021-01-20:
I have both compilers. The Mikro compiler did not flag it. The MpLab one did flag it. Like I said I stripped out most of the things I do not need and just using it to to match the antenna, Took me awhile to get my head back on straight and relearn lots of stuff I forgot. I've been retired for 16 years. Like I said if you want what I have or end up with you are more then welcome. I like the MCC function of the MpLab. Thank again.

actionwatcher, 2021-01-22:
@sparkey2450: Steve, do you have MPLab project shared somewhere?

Thanks, Arkady/NU6N

sparkey2450, 2021-01-24:
David,
In the routines get_reverse() and get_forward() you setup FVRCON for 3 different reference voltages for the ADC and return based on a value read. What was the reason you used all 3? FVRCON - 0b10000001, 0b10000010, and 0b1000011.  Thanks, 
Steve W8TTY

Dfinitski, 2021-01-24:
Hello, Steve.
That is increasing a measurement accuracy.

sparkey2450, 2021-01-25:
David.
Still trying to get my head rapped around the use of variable Vrefs in get_forward() and get_reverse(). Could not the forward measurement and the reverse measurement end up using different Vref's in their reading during a calculation cycle? Maybe it doesn't matter. Anyway I'll get it eventually. My real question is where did all the correction numbers come from in the conversion() routine? Hopefully my 2nd tuner will be here this week and I will get a chance to start testing the software.  Hooking up the PicKit3 for debug is just another hurdle. I appreciate you taking the time to answer my questions. Thank You
Steve W8TTY

Dfinitski, 2021-01-25:
Hi Steve
I don't understand you, where did you find the Vref variable ? I don't see
it.
Numbers in the correction routine are results of real measurements. Those
do compensate for diode's characteristic and make power measurement is more
accurate.

sparkey2450, 2021-01-26:
David,
I figured it out as to what was in conversion(). The Vref was where you set it up in FVRCON bits 0 and 1 where you select the different Vrefs in the ADC. All is good now

g8kwx, 2021-02-17:
Has anyone sucessfully ported the source code project from MikroC to MPLAB or other IDE ?

sparkey2450, 2021-02-17:
G8KWX,
I've got some of David's Code running on MpLap's IDE and XC8 Compiler. I have a version that does tune. I stripped out all the code that did not pertain to the box I was using as it was prebuilt. I also removed the OLED as I'm matting this to a Raspberry Pi to use the WiFi. Its far from done but you are welcome to where I've taken it so far. I've gone through much of the MikroC stuff and converted it to MpLabs. There are a number of function calls used that do not port over and are unique to MikroC. I used MCC to generated the Init code.  If you want a copy of all of it you can just go to my QRZ page and use that email to make a request and I will send a zipped up bundle of the MpLabs Project. I have taken David's coded box, set it to Auto, and stuck it out at the antenna. It works well that way. And if that what you're looking for there's no need to change anything. 

Steve (W8TTY)

khkremer, 2021-03-08:
Steve,

when you scroll up to the beginning of this thread, you will find that I started this ;) My PIC skills were not good enough to start port, but I might be able to work with what you already have done, and would not mind a copy of your code s as well. 

Thanks and 73,

Karl Heinz - K5KHK

sparkey2450, 2021-03-08:
Hello Karl,
Yea never know with email how things go. This is the first email I've seen from you. Here is a copy of what I sent. Its know where near done but a start. It does do tuning but sometimes misses the mark. When you compile this with MpLabs XC8 compiler you will see a bunch of warnings about functions that are not called.  I used MCC to generate much of the code to the PIC. They have everything in there but the kitchen sink. There is no OLED interface with this code and I stripped out anything I did not need for my particular box.  Also there are some of the original c code functions that are no longer used. I found it hard to follow the original so I just started to type in my own. Been a while since I looked at this as I've been working on the I2C communication stuff and not the tuning. I figured I'd get back to tuning when I get the communications running.This is a start and gets it compiling under MpLabs. Originally I was going to hook this to a Raspberry Pi, which I do have running on some other code. But after getting into it, it was clear that what was really needed was hooking a WiFi module to the tuner and having the Tuner do the communications. I'm waiting on that module now. Much of the I2C stuff with Raspberry Pi I did should carry over. None of that is in what I'm sending. I tried to put in more comments and make the code more readable so I hope that helps. Good luck and have fun. This is a good board to learn programming on.   

Steve (W8TTY)

g8kwx, 2021-03-08:
Steve, was there an attachment somewhere ?

sparkey2450, 2021-03-08:
You will not get it unless you email me directly. My email is on my qr page for W8TTY.

g8kwx, 2021-03-08:
Steve, as said I have emailed you several times. Just now also. Maybe i am going to junk :-)

davidhay45, 2022-01-25:
@sparkey2450 how does your project stand at the moment?
I have just (today) started to compile the code using MikroC 7.6 and obviously the learning curve is high.  Any pointers, like which libraries do I need?

Dfinitski, 2022-01-25:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/12/1021397701@

Hi. I will try new matching algorithm soon.
All needed libraries are included

sparkey2450, 2022-01-26:
I'm not using MikroC. I used MPLabs. I sort of shelved the Tuner Software as I got involved with Mag Loops. Been so long since I looked at the Tuner software I'd have to almost start all over. I have a Mag Loop Tuner that uses WiFi. Its developed on the ESP8266 Module and a stepper motor. The Capacitve tuning for the Mag Loop is complete. I've had the Mag Loop up and running. The human interface is Blynk. I use the local server version.  There are some pics on my QRZ page. I'm in the process of also setting up a WiFi control for an Antenna Rotor so I can rotate the Mag Loop. The Rotor software is about 75% complete. I just need to come up with a way to indicate direction. The end result is a mag loop that can be tuned and rotated from the shack using WiFi. The only thing else needed is 12 VDC to power the equipment at the antenna. Battery works well mated to a solar panel charger or 12 VDC run over the coax.  Steve (W8TTY)

davidhay45, 2022-01-26:
@sparkey2450 Thanks, I'll comment here as to how I get on.
@Dfinitski Thanks for coming back. I am hoping to make a version of the software which allows a small number of pre-tuned settings to be used with WSPR on different bands.  I have not got a test bed yet but my experiments with MiKroC 7.6 look good.  One question, do I need to change the config registers externally, at the moments I have
CONFIG1 : $8007 : 0x3FEA
CONFIG2 : $8008 : 0x1633
Thank you for making this software available!!

Dfinitski, 2022-01-26:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/12/1022225507@

Hi. All the configurations parameters set with the MikroC. See the project
menu.

sparkey2450, 2022-01-26:
The best place I ever found to put the ATU100 is out at the antenna feeding a 123 foot long wire with a 9:1 UNUN. I used David's earlier software build 3.0 and put the ATU100 in Auto out at the antenna. It did make a difference. I could work 160 thru 10 meters. I got the code ported over to MPLabs Compilier but rewrote much of the MikroC because some of the Libraries and functions used by MikroC are not transportable. I used a highly modified I2C sort of protocol to talk between the ATU100 and the ESP8266. The ESP8266 was setup as a server so you could get at the interface at the antenna via a simple web browser over WiFi. I had the proof of concept software completed and just needed to do some work to make it more reliable. Was not happy with the tuning portion. That's is when I got interested in looking into a Mag Loop. Very Impressed with that antenna. I hear things and make QSO's that were impossible on the long wire. Mated it to a Hermes Direct Conversion SDR and the reception runs circles around my ICOM 7100  . Biggest draw back of the Mag Loop was the tuning. Its extreme narrow bandwidth characteristic which makes it quiet is also one of the drawback if you like to surf the band you are using. This is why I did the WiFi interface to the tuning capacitor. The antenna works just as well as any dipole I've used and it only needs to be a few feet off the ground.
 
The presets you talked about is a good idea. Especially for some of the communication protocols that are at predefined frequencies. Good luck on your project. Maybe someday I'll get back to the ATU100.