Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #13: Relay S8 appears to be energized when the ATU is in its Reset state

rogerclarkmelbourne, 2020-09-05:
Hi.

I've noticed that I relay S8 seems to be energized when the ATU is in its Reset state.

As far as I can tell, S8 appears to connect the capacitive load chain to either the RF input or RF output side of the inductor chain, but in the Reset condition.

Was there a reason for doing this, as it causes the ATU to draw some unnecessary current when its in its Reset state.

Dfinitski, 2020-09-05:
Hello. Yes it is.
Because in that relay position the own SWR much better due to compensation own inductance and capacitance.

rogerclarkmelbourne, 2020-09-05:
If the PCB was redesigned, could this be resolved by changing the connections to the relay, so that the Normally Open and Normally Closed were transposed ?

Dfinitski, 2020-09-06:
See the schematic, the relays contacts are shown in the power off position. You will see how the inductances are conected to input or output when the relay is in ON position.

rogerclarkmelbourne, 2020-09-06:
@Dfinitski 

OK.

I viewed the S8 as selecting which side of the inductors the capacitors are connected.

As far as I can see, one side of inductor L1 is connected to the RF input J2, and one side of inductor L7 is always connected to the RF output.

i.e 

```
RFin --- L1 --- L2 --- L3 --- L4 --- L5 --- L6 --- L7 --- RFout
         |                                              |
         -------------------> | <------------------------
                              |
                              |

                           C1 to C7
```

closed 2020-09-07 by rogerclarkmelbourne