Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #15: Simple Test mode - Inductance stepping

va3rr, 2020-09-08:
David - my recently constructed ATU-100 EXT seems to work as intended.  As well, the Fast Test Mode energizes all the relays as expected.

However, in the Simple Test Mode, I cannot step through the entire range of inductances, as expected.  Inductance is added as expected until reaching 2.20 uH on the LCD, then all the inductances are removed, and the cycle repeats until 4.50 uH, when all the inductances are again removed.  It's not until I reach 6.70 uH on the LCD meter that expected amount of inductance is added, and works normally until 8.53 uH.

I checked the capacitance side, but all the relays seem to add capacitance as required, when stepping through using the Simple Test Mode.

Is this an issue that may affect the normal functioning of the ATU-100, or should I not worry about it?

спаси́бо de va3rr

Dfinitski, 2020-09-09:
Hello.
What kind of display do you use ? I have tryed with OLED 128*32 - works good.

va3rr, 2020-09-09:
I'm using a 1602 display and it works well.  It's just that the relays (just for the inductance) do not energize when expected in the Simple Test Mode.

All of the relays work, but not in correct order.  When adding inductance, everything works fine and increments normally up to 1.83 uH and relays are 0011111. (where 4.4 uH is most significant digit)

But when I step up to 2.2 uH, relays are 0000000 and when I increment once to 2.25 uH, relays are 0000001, when I expected 0100001.

Relays for 4.4 uH and 2.2 are never energized in Simple Test Mode until I reach 6.70 uH, then all relays are energized as expected (1100000), and all relays energize as expected in proper order right up to 8.53 uH (1111111).

My apologies if my explanation is not too clear...

Dfinitski, 2020-09-09:
Other words, A5 line of PIC or 2.2uH control line is not working.
Please check again this line in Fast Test very carefully again.
Use a multimeter to measure a voltage between relay's coil nodes. 
I have check the code, no errors here.

Dfinitski, 2020-09-09:
Here is the code, L_mult = 4:
```
if(L & ind<32*L_mult-1) {
              ind ++;
              set_ind(ind);
           }
           else if(!L & cap<32*L_mult-1) {
              cap ++;
              set_cap(cap);
           }
```

va3rr, 2020-09-09:
Well, you are indeed the master of your wheelhouse.  Thank you. My supplier in Irkutsk had left an odd combination of cold solder joints and a perhaps a solder bridge (to VSS?).

What was confusing me was that in Fast Test Mode, somehow 4.85 volts WAS being applied to the gates of Q12 and Q13.   But in Simple Test Mode, nothing was being applied to those two gates, except in the circumstances I described above.  In fact, the two gates had 0.65 volts on them when they should been very close to, or at 0 volts, when not active.  Perhaps because RA4 and RA5 are physically close to VSS?  IDK...

If I were to build this kit again, I would certainly recommend that prospective builders put a voltmeter to the gate solder pads for all the mosfets BEFORE soldering them, using Fast Test Mode.

In any case, all is working as designed.  Again, many thanks!

closed 2020-09-09 by va3rr