Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #16: Nothing during TUNE

Jean-france, 2020-09-09:
Hi,
I bought the full installed ATU  (CGJ-100), see picture.
The firmware is V3.0.
There is a problem because when the ATU isin AUTO mode, we can see SWR : TUNE but nothing happen. ATU stay still or until i BOT it. So no according antenna do.
Can you explain what is this issue ?
Many thanks in advance.
Best regards,
Jean

Dfinitski, 2020-09-09:
Hi. 5 Watt is not enough for normal operation.
You have 2 ways to resolve this.
1) reprogram cells to work starting 1 Watt
2) rebuild the tandem match to 1:5 and reprogram cells.
Second is much better.

Jean-france, 2020-09-09:
Hi,

Many thanks for your reply.

OK, it’s working but with 28 W output power from TX (with linear amplificator).

Sorry, but how to rebuild the tandem match to 1:5 and reprogram cells ?

BR
Jean

Dfinitski, 2020-09-09:
Now the tandem match has 10 turns, remove 5 and leave 5.
After that buy  Pickit 3 programmer and change cell 05 to 01 value, cell 31 to 05 value.
Read the user manual you can find here. Goggle "pickit 3 programmer how to".

Dfinitski, 2020-09-09:
Read this conversation also
https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/2#issuecomment-499742247