Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #17: T1 & T2 turns Ratio formula

KenLotts, 2020-09-10:
Is there a formula to use to design a custom T1/T2 turns ratio for optimum use at a given power level?

I see the manual gives a solution for 40 watts (1:5) and 150 watts (1:10) and for some reason they even mention 1500 watts (1:32)

The manual mentions that the turns ratio should be designed to prevent the sense voltage from exceeding 4.096 Volts for the PIC16F1938 processor and 5.0 Volts for the PIC18F2520 processor."

So ...  I was wondering ...What is the formula I would use to optimize T1/T2 for a given maximum power level?

Ken Lotts

Dfinitski, 2020-09-10:
Hello, Ken
That is formula based on Ohm Law, I hope you heard something about that.    I = U / R.      P = U * I  = U * U / R.
So average voltage value for an example 300 Watt of power equal to U = SQRT(P * R) = 122.5 V. It will be decreased on a tandem match according to his ratio, an example 1:20, 122.5 / 20 = 6.12 V . Actually this voltage will be closer to peak value, so multiple it to SQRT(2) = 1.41,   so 6.12 * 1.41 = 8.65. This voltage will be devided by 3 with the resistor devider, so 8.65 / 3 = 2.88 Volts.  A you can see this value is pretty far to 4.1 and 5 Volts, so this tandem match ratio is to big for maximum power 300 Watt.

KenLotts, 2020-09-11:
Thanks.. Good job.  Nice explanation, I had neglected the 3:1 divider which is what threw me. 
And the half wave rectifier was the final piece... plus a bit of efficiency fudge
So..  I wrote an approximate formula that seems to work (in excel) .. I made a fudge guess of subtracting 0.27 to the solution since that made it line up better with the provided examples of 40, 150 and 1500 watts

Optimum Turns Ratio = ((SQRT(<MAXPOWER>*50)*SQRT(2))/12)-0.27

I checked the formula on the examples and got:

40 watts = 5.000463 turns ratio
150 watts = 9.936207 turns ratio
1500 watts = 32.00486 turns ratio

Ken
AA7JC
--

--

--

rogerclarkmelbourne, 2020-09-11:
Guys. 

Initially this seemed counter intuitive to me, but I see that effectively T1 is the primary winding and the unlabelled straight through line is the secondary. Hence the pickup's are currently a 10:1 step down transformer not step up.

So using more turns on the primary (T1), increases the step down amount.

Is this logic correct ?

Dfinitski, 2020-09-11:
Excellent, Ken
It is very helpfull for people.
But where is a power in this formela ? 
**Optimum Turns Ratio = ((SQRT(*50)*SQRT(2))/12)-0.27**

Roger, that is step down transformer, correct.

KenLotts, 2020-09-11:
Sorry.. Cut and paste did some funny business

Optimum Turns Ratio=((SQRT(MAXPWR*50)*SQRT(2))/12)-0.27

rogerclarkmelbourne, 2020-09-11:
@Dfinitski 

Thanks for the confirmation.