Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #18: Line of code that appears to do nothing

rogerclarkmelbourne, 2020-09-15:
Hi

Reading your code. I noticed this line in the tune() function

`   SWR - get_swr();`

https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/blob/master/ATU_100_EXT_board/FirmWare_PIC16F1938/1938_EXT_board_sources_V_3.0/main_main.h#L471

Although this probably compiles, it does not seem to do anything apart from running the get_swr() function, as the result of the calculation does not seem to be assigned to a variable.

Dfinitski, 2020-09-15:
Hello, Roger
Thanks for this notice.
If the compiler was rejecting this line - this is error.
If get_swr() is running - this is not.
Depending to compiler. Usually I get warning message about nonused variables and something about this.

Dfinitski, 2020-09-30:
Hi, Roger
Where did you find the "main_main.h"  ???
Only "main.h" is present with different code, without that error.
[](https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/tree/master/ATU_100_EXT_board/FirmWare_PIC16F1938/1938_EXT_board_sources_V_3.0)

> Also. The get_swr() function seems to return the global value SWR, which is pointless as SWR is global
It is not pointless when you write "if(getswr() <=120)"

rogerclarkmelbourne, 2020-09-30:
Hi David

Main_main.h is in

ATU_100_EXT_board/FirmWare_PIC16F1938/1938_EXT_board_sources_V_3.0/

Re: pointless returning SWR

Yes. I see what you mean.

However it would probably be better use the return value of getSWR everywhere and not have getSWR update the SWR global variable.

i.e you can change

`if(getswr() <=120)"`

to 
```
getswr();

if(SWR <=120)
```

Personally I'd be vary wary of returning global values as its a potential source of hard to find bugs.

But I guess your codebase isn't very large, so its no a big problem

Dfinitski, 2020-10-01:
That's interesting , I don't see that file in my browser. Also there is not exist such file in my project. Main.h is exist, main_main.h is not.
Nevertheless I am able to open it by link you provided.... So i am interested how did you find that file ? It looks like some old version code or something like that.

rogerclarkmelbourne, 2020-10-01:
When I cloned your repo onto my local machine it brought down that file

I double checked and there are 4 copies of the files on my machine, one in each of the version 2.9 and 3.0 and subfolders for the 2 PIC variables i.e 2 x 2  = 4

Is your local machine in sync with what's on github

Did you try 

git status

or git push

I'd advise you to copy the folder before doing anything, as it sounds like whats on your machine is not what's in github.

Dfinitski, 2020-10-01:
I am sorry, that was some junk file, it is not using for compilation. Main.h is the correct main file.

rogerclarkmelbourne, 2020-10-01:
No worries...