Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #20: PICkit 3 tips

3d0g, 2020-09-23:
Been a long time since I messed with PIC micros. Didn't have any fancy gui tools back then. In PICkit 3 I can read and verify the PIC16F1938 fine, check communication says PIC device found, but when I try to write the firmware hex file, I get a "Programming failed at Program Memory address 0x000001." Re-reading the PIC shows that only the 1st word was written to Program Memory and 1st byte to EEPROM Data.

What am I doing wrong?

Dfinitski, 2020-09-23:
I saw the similar error  when my Pickit3 programmer was burned. I had
burned him when Iwas playing with a tuner board, some bad power sequence or
something like this, I did not remember.
Just ordered a new one.

3d0g, 2020-09-23:
It's a brand-new PICkit 3. Just unboxed it today. I rummage through the parts bins and see if I've got a DIP PIC somewhere to test the PICkit with. SSOP PICs are brutal on these old eyes.

Dfinitski, 2020-09-23:
It does not matter what kind of package you are programming. You can read
the chip, so all the connections are good., but for successful writing, the
programmer should generate relatively high voltage on the MCLR pin, about 8
Volts. If you have an oscilloscope, you can check this.
Also you can try to use the programmer in a LVP mode, search the options.
In this case a MCLR voltage is not exceeding 5 Volts.

rogerclarkmelbourne, 2020-09-24:
BTW. I have an old PICKit2.  Would this work ? I think the PIC18F1938 is supported.

Dfinitski, 2020-09-24:
Hi, Roger
It is not supported but you can find in the Internet updated file for new
PIC support.

rogerclarkmelbourne, 2020-09-24:
Thanks David,
I'm a bit cautious about attempting to even change the EEPROM, but I have another kit on order from China, so will probably do some serious hacking on that one when it arrives

Dfinitski, 2020-09-24:
I have ordered Microchip Snap board - very cheap and compatible with all
the new PICs. But only LVP programming.
Rapidly fast delivering from Microchip Direct Store, 2 days.

rogerclarkmelbourne, 2020-09-24:
Thanks David.

I'm actually in the process of porting your board to use a STM32F103 micro-controller, as I"m more used to using ARM MCUs. 

However as you will appreciate, its not a 5 minute job, as I also need to layout the board again, using KiCad, as well as port the firmware... So I estimate it will be Christmas before I have anything vaguely working

Dfinitski, 2020-09-24:
Why dont you use PIC32 ? Those are ARM too :)

rogerclarkmelbourne, 2020-09-24:
I'm most comfortable with STM32 MCU's.

3d0g, 2020-09-26:
OK, I got absolutely nowhere with PICkit 3 v3.01 software, every PIC variant I had in the bins failed. So I ordered another programmer but while waiting I figured I'd try MPLAB X today. Reverted the programmer back to MPLAB mode and lo and behold, MPLAB IPE programs up the PIC just fine. Tuner is alive and kicking!

Now I remember why I quit using Microchip in my projects. ;-)
Thanks for the assist folks.

closed 2020-09-26 by 3d0g

Dfinitski, 2020-09-27:
That the reason why Pickit 3.xx is no longer supported by Microchip and it is impossible to get it from offciall Microchip's resources.

3d0g, 2020-10-03:
Just to finally close the loop on this for others that may be reading, I finally got in a GENUINE PICkit 3 from eBay and it worked flawlessly with the PICkit programming software. My original programmer was clearly a chinese knock-off. The board looks completely different than the Microchip one, even though the case is identical. Lesson learned.

MOUSE1025, 2020-10-04:
Dave,
Thanks for the heads up. I am going to check mine and see if it is a
knock-off.
Kind regards,
Christopher

kk4das, 2021-11-19:
I know this is an older thread but I experienced a similar problem.

 I just completed my tuner build and wanted to update to the 3.2 firmware.  I have never used anything PIC before but I'm a 40 year plus software guy so "How hard can it be?"   :) 

I'm using a PICkit 3 clone randomly chosen from the dozens on the Bay.  Downloaded the 3.01 programming software from the Microchip archive.  Every attempt at a write fails on verify and similar to the OP - a read appears to have only written the first byte or perhaps some random bytes.  That said, after multiple attempts at writes I popped the display back on and the doggone thing booted up in the 3.2 firmware and as far as I can tell it works flawlessly. 

I'm probably going to lead a Maker's group build of this tuner so I'd need to be able to reliably work on it.  Don't know if there is a knock off programmer vendor that is more reliable or if there is another approach I could try.  "Geniune" PICkit 3's go for ridiculous prices on the Bay.  PICKit 4 is $70 direct from Microchip so maybe that is the way to go.

Any suggestions would be appreciated.

Thanks and 73,
Dean
KK4DAS

Dfinitski, 2021-11-20:
Hi.
Chinese clone of Pickit 3 works great and has a reasonable price

3d0g, 2021-11-20:
The MPLAB IPE works fine with the PICKit clones in my experience. It's just the PICKit 3 software that has issues.