Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #21: Someone asked me to find out if the MCU wakes periodically to check the SWR?

rogerclarkmelbourne, 2020-10-07:
Hi

I've not noticed this myself, as my local QRM is too noisy, but someone who uses your ATU for SOTA work, reported to be that they thought the MCU was occasionally waking up and causing QRM

I've had a look in the code, but I can't see anywhere that the MCU is put it a sleep mode, but its quite possible you are doing this via a direct register setting to the PIC and I'd not notice that in the code.

Dfinitski, 2020-10-08:
Hello, Roger.
Some people use chinese full assempled tuner with a built-in Li-Ion battery. This device containes a power bank standart electronic board with a DC/DC PMIC wich is the source of RF interferences.

I never heard before from hundreds russians and polish users about this problem. They use this autotuner 2 full years.
It was first published in 09/09/2018 on cqham.ru

The MCU can not to go sleep because it meassure a power and SWR continiously.

rogerclarkmelbourne, 2020-10-08:
Hi David

Thanks for the quick response.

I'm afraid I don't know whether these SOTA guys were using a chinese kit with a built in battery or not.

But, I'll pass the message back to the guy who asked me, (who wasn't the person who directly experienced the problem)

Also, its interesting that the MCU is constantly sampling the SWR.
As the MCU seems to be taking virtually no current as far as I could tell, I presumed that you put it into a sleep mode, and only woke it up when a button was pressed.

But I suspect my cheap Chinese made bench PSU, fails to register any current when the board is only taking a few mA

Thanks again for the reply

sandroatu100, 2020-11-13:
Hello
I connected a 50 ohm load (Bird 3 Ghz, but is the same with any load) to the out port, Bypass function activated.
The swr indication with 10W( any power) is:
7 MHz swr 1.2
14 MHz swr 1.5
24 MHz swr 2.1
30 MHz swr 3.0
above higher
But really on the at-100 input there is 50 ohm tested with nanovna and swr meter
May you reply a hint?
Atu-100 is 7 cells version with firmware 3 and oled display
Regards
Sandro

Dfinitski, 2020-11-13:
Hi.
If you have a NanoVNA, you can connect it to the tuner input and check the SWR by frequency and even find why the SWR is so large.
I am pretty sure that the large relay's capacitance is the reason to that.

Dfinitski, 2020-11-13:
Here is a other user report, bypass is above
http://www.cqham.ru/forum/attachment.php?attachmentid=307210&d=1550015243

sandroatu100, 2020-11-14:
Hi David,
I tested with nanovna the atu-100  input and output swr with 50 ohm load and is correct, that means no mismatch is caused by the atu100.
    It is wrong  the indication on the atu100 display coming from the internal directional coupler made with ferrite.
(on Ebay there are two versions with short or long ferrite, mine is long)
Attached the photos
[image]
[image]
Tested the 50 ohms resistors smd on the pcb are correct.
Sandro

sandroatu100, 2020-11-14:
Hi David,
I found where is the problem.
In bypass mode there is a single pole relay but should be double pole.
The through path goes by the inductance path all shorted but the path is too long.
I shorted as a test with a wire bridge the path and now the swr reading is correct.. so will put a double pole relay and modify it.
Photo is attached.
[image]
Thank you for this project, let me know your opinion.
Regards
Sandro

Dfinitski, 2020-11-14:
Hi, Sandro
I think you are not right.
Check the schematic, no any double pole relays there.
As I told you before, if you have a NanoVNA, connect it to the input and
find what is the reason for the large SWR.

sandroatu100, 2020-11-15:
Hi David,
yes I know there is only 1 relay , I have the schematic.
Using a double ralay , in bypass mode, is better the isolation from the atu to the antenna and so the swr(casued by the atu)
It's an improvement in bypass mode and also other network analyzer confirmed. But is possible to use as original.
 The atu works good except one condition:
Frequency 3,77 MHz antenna is a loaded dipole
In manual mode I found perfect matching with 4,95 uH and 0pF .   But pushing tune button no way to match the load, the
controller don't find the tune automatic. I can do only manually.
Anyway, I appreciated the useful functions in most cases.
Regards,
sandro

Dfinitski, 2020-11-15:
Hi, Sandro
There is no any special signal from mcu for bypass mode.