Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #22: reprogramming atu-100

olson339, 2020-10-30:
Need some help with the i2c reprogramming. I have a 1602 LCD with a i2c backpack. I used i2c scanner to find out the address, which is 3F. I changed the address using a PIC 3 and wrote the change to my ATU-100. Connected the LCD, background light comes on along with 16 square blocks on second row. I readjust the brightness but no readable display. I am using ver 3.0 of your frimware

Dfinitski, 2020-10-30:
Hi.
Your I2C address scanner gives you 7 bit address, but the tuner board uses 8 bit address. 
As you can reed in the manual,
"**The port extension board for display 1602 made on the PCF8574AT chip has the address 7E, on the chip with the “T” index - 4E.** "
So no any ambivalent understanding which address you have to use.

olson339, 2020-10-30:
so the scanner shows 0x3F. I enter 3F in cell 00 of the EEPROM. Should the entry be different

Dfinitski, 2020-10-30:
Did you read my previous message ?
What kind of chip is used in the "backpack" of LCD ?

olson339, 2020-10-30:
yes I read your previous message - the chip on the backpack is PCF8574AT

Dfinitski, 2020-10-30:
Excellent, your I2C address is 7E.

olson339, 2020-10-30:
ok thanks

olson339, 2020-10-30:
The tuner is working but I haven't added any buttons yet for auto/bypass

Dfinitski, 2020-10-30:
Can you confirm this behavior ?
https://www.youtube.com/watch?v=mUfD4gnzswc&t=1s

olson339, 2020-10-30:
I ran the tuner using my 7300. Display worked correctly from star, tune, operate

Dfinitski, 2020-10-30:
Thank you for the help.
you can test the newest firmware https://drive.google.com/file/d/1sCYbo3DPGjM-r94aqZJpuZqGIvasgv6K/view
if you want.

valantis-chios, 2020-11-04:
Hi, I would like to inform you that the DIY kit that I bought has the same problem as described in the video. Firmware is 3.0. what do you propose to do? I am using lcd 16x2. I am waiting to receive pic programmer to check the hex and eeprom.
Thanks in advance

Dfinitski, 2020-11-04:
Hi, 
the 3.1 firmware version should work fine. Use a Pickit 3 to change the old firmware.
If you will have some problems with 1602 LCD please tell me know ASAP.

valantis-chios, 2020-11-13:
20201113_161618.mp4
<https://drive.google.com/file/d/1aXIEYz6RTYQtXtyHNyLcT5btIjO8TpyP/view?usp=drivesdk>

  20201113_161849.mp4
<https://drive.google.com/file/d/1elP15rnoYLPoujuj1u5GEO6MsJs92JkB/view?usp=drivesdk>

Dfinitski, 2020-11-13:
Hi,
What kind of chip is used in the I2C extender of your LCD display ?

valantis-chios, 2020-11-13:
Hi, the chip is PCF8574T

Dfinitski, 2020-11-13:
Thanks.
I will order LCD with this chip and check this by myself.
I can not find the problem in the code also other users report about good
work with LCD.

valantis-chios, 2020-11-13:
One more thing,  how does the type of display affect the tuning process?

Dfinitski, 2020-11-13:
I is not matter, you can use the tuner even with no any displays, it is
only an option.

valantis-chios, 2020-11-13:
20201113_235455.mp4
<https://drive.google.com/file/d/1IMidnI6zYtDpbtd9VnnCxdzalz9Px1o9/view?usp=drivesdk>

valantis-chios, 2020-11-13:
Hi, I rewind tandem match and the problem with tuning seems to be solved ( maybe there was a short between filament and ferrite). But the problem with no swr indication  still exist.

Dfinitski, 2020-11-13:
The LCD will come to me next Thursday. I will see.

Dfinitski, 2020-11-17:
Hi.
I have bouth randomly an LCD with PCB8574T expansion board on Ebay, I2C address is 4E.
Everithing works good. 
I tried to find mistakes in the code also, many times, but I don't see anything wrong.

Dfinitski, 2020-12-01:
People, watch this video https://www.youtube.com/watch?v=vWtbs4n4ewo&t=18s
If somebody knows what is going on - tell me know, I will fix it.

valantis-chios, 2020-12-01:
Hi, I have the same problem like no 2 and no 3 lcd( no swr indication). Tuning possess is ok.

Dfinitski, 2020-12-02:
Hi. Order the chiepest LCD on Ebay or Aliexpress, should to work. I don;t know what is the reason, so I can't resolve this problem. 
I don't have a strong expirience with this kind LCD, i have no idea what is going on. But definatelly those LCD have some differences.

owenaggi, 2021-02-26:
help i dont understand what i have done i have the picket 3 and ext board i wanted to use it on low power when trying to reprogram i have some how deletet the pic and now when i try to program the pic picket 3 tells me that there is insuffichet memory please tell me a what i did wrong and if possable how to fix it

TA2KW, 2021-10-16:
hello, I made a connection with 1602 LCD, I2C PCF8574T converter, but the 7805 transistor in the circuit supply is overheating, I could not solve this problem, why does the 7805 get hot? :(

Dfinitski, 2021-10-16:
Hi.
Everything works good but 7805 is hot ?
What is the voltage rate of the relays ?

TA2KW, 2021-10-16:
1602 LCD screen gets hot when I plug it in,
No problem with oled display.

Dfinitski, 2021-10-16:
So check the connection, power voltage polarity.