Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #23: Replacement Diodes

KevAmiga, 2020-11-05:
Afternoon all - I need to replace 2 of the diodes on the underside of the board, but am unsure of the values, or in essence what diodes to purchase. Installing them shouldn't be a problem. This issue was down to my mistake, i needed to modify the case and slipped with the tool hitting the components (lesson learned!)

Please excuse the rubbish photo, it was the best i could get .

[image]

Hoping someone can point me in the right direction.

Many thanks

Kev

Dfinitski, 2020-11-05:
Hi, Kevin
According to the schematic those are BAT41 diodes.
This info is not a secret, everyone can see the schematic.

KevAmiga, 2020-11-05:
Excellent - thanks, have found the schematic will take a look.

Kev

closed 2020-11-06 by KevAmiga