Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #25: Updating an ATU-100 EXT

alexander-pick, 2020-11-18:
I have got an ATU-100 EXT which is a china build. Trying to update it via the connector seems to fail. The MATLAB X IPE 5.40 gives false positive "programming ok" messages but seems to write nothing and the PICKit3 v3.10 standalone flasher is unable to detect the device. Reading works on none of the both solutions. I try to power the device via the programmer. Is there any hint on the settings required to make it work?

Dfinitski, 2020-11-18:
Hi.
Yes. Try to choose Midrange -> 1.8 Volt MCU core in the PICKit3.10 first.
It is impossible to get false positive message, because a program check written memory after programming.

alexander-pick, 2020-11-18:
Thanks, that was fast! Got it to working and flashed the new 3.1 firmware. Than I did the patches for QRP from the other thread:

simple mode (30=00)
tandem match ratio 1:5 (31=05)
Min. start up power 1W (05=01)
upper power limit 40W (06=40).

This causes my W display to be of by the ratio 1:4? It shows 2.5W instead of 10W etc. Without the changes and on 3.0 it worked fine. Any idea what is wrong?

alexander-pick, 2020-11-18:
Ok, this seems to be based on the tandem ratio. If I change this field the W display is wrong.

Dfinitski, 2020-11-18:
Did you change the tandem-match ratio ? I mean physically, reducing turns to 5 for both sides ?

alexander-pick, 2020-11-18:
What do you mean exactly? So far I just edited the values above in the eeprom. Is anything else (physically) required too?

Dfinitski, 2020-11-18:
See the video in the readme file.
Of course, you must reduce turns on the tundem-match. After that you can indicate correct ration in the firmware for correct power calculation.
What you are doing now just changes calculation formula in the firmware.

sandroatu100, 2020-11-27:
Hi,
I see in the datasheet PIC16F has the sleep mode.
Is it possible to modify the firmware to include this feature?May be with resume at one keystroke.
This would be great for power saving in the field use and, to cancel the rfi irradiation by the cpu.

Dfinitski, 2020-11-27:
Hi.
No sense to save 3mA of CPU power consumption when each relay takes 15 mA.
And your radio also takes 200 - 500 mA in RX mode.

sandroatu100, 2020-11-28:
My radio in Rx mode is less of 30 mA and not all the relays may be activated.Also I want to power off the display when not in use,this make sense for Qrp continuos reception.
But, more important, I measured the interference irradiation by the cpu and is very high and goes in the input of the receiver!!
Some digital antenna tuners have this sleep mode and the sleep instruction on the datasheet is easy( page 354).

Dfinitski, 2020-12-01:
Hi. You can power off the display right now, see the manual, cell 32.
What about a sleep mode, it looks like possible, but I am not sure about working algorithm. If the MCU will be in a sleep mode, it could not meassure the input power more, so what criterias to wake up for it ?
One idea looks better - turn on the relays only when input power detected. In this case you could save much more energy while you stay in RX . But in this case MCU should be always in working mode.

sandroatu100, 2020-12-01:
Hi, 
ok for the display off.
The criteria may be: after 10 seconds of no operation mcu goes to sleep, than wake up with an interrupt function from pushbutton  ,   but I am not expert with software ( just hardware engineer) .
The sleep also avoid interference from mcu to the receiver , I measured S3-S7 in the receiver depending of frequency.
Rfi is the main issue, the save energy it's also nice.

Dfinitski, 2020-12-02:
I think that is bad idea, every time when you need to change a band click a button ? No, tuner must to work in automatic mode.
I dont agree with you about interferences. I will check it soon and give you result. You can try also with disconected display, probably it is a souce of them.

sandroatu100, 2020-12-02:
The interference come from the cable from mcu to display. I tried to put a ferrite bead and short cable, it's better, but with a receiver sensitivity better of -130 dBm is not so easy to cancel rfi, that's the reason for my question to sleep the mcu.
 I have to try by myself an hardware solution, will keep you informed.

Dfinitski, 2020-12-03:
Sandro, a coaxial fider is not working like antenna, I need more info about your setup to make a conclusion, but I never heard before about RFI from this tuner.
If it will be helpfull, you can use the tuner without any display.

Dfinitski, 2020-12-31:
Hello, Sandro.
Please try the 3.2 version. I have resolved countinious data transfer to a display problem. You can also turn off the display by timer in RX mode, also with all the relays for your choice.

sandroatu100, 2021-01-01:
Hello David.
Thank you for your support. I will try the new version just I'll receive my new pickit 3 programmer.
Happy new year 2021 and stay healty.

PY2BND, 2021-01-10:
> y the 3.2 version. I have resolved countinious data transfer to a display problem. You can also turn off the display by timer in RX mode, also with all the relays for your choice.

Thanks - Very nice feature!!! 73!!