Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #27: Inductance toroid cores

ON6NT, 2020-12-13:
Good morning
When I check the number of turns through the T68-2 toroids, I never arrive at the inductance values, stated in the schematic. T68-2 cores have an AL of 57 uH/per 100 windings. For L4 resp. L5 for instance 8 resp. 13 turns are given. For L4 this ends up in 5,6 uH in stead of 0,45 uH from the schematic; for L5 we find 7,7 uH in stead of 1 uH. The same goes for L6 and L7. Is the schematic wrong, or are the cores not T68-2 ?

ON6NT

Dfinitski, 2020-12-13:
Yes, robably this project is wrong........
Oh, no, that is your calculation is wrong ! Whoa!
http://toroids.info/T68-2.php

ON6NT, 2020-12-14:
Hello David
Thanks for the reply. All clear now, I was confused because AL has uH/100 turns as a unit.
Thanks !