Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #28: Difference between PIC16F1938  and PIC18F2520 processors implementation

IK2HEW, 2020-12-17:
Hi all,

I'm really sorry but looking on search engines I wasn't able to find any information about the difference on the two processors implementation. I'm certain that somewhere there will surely be a lot of discussions on this matter but I didn't find them. Please, offering in advance my apologies, is anybody here that can tell me something about ?

Thanks all in advance

Pino, IK2HEW

Dfinitski, 2020-12-18:
Hi.
PIC18F2520 is a pin to pin compartible to PIC16F1938. It is pretty old MCU and it is much easier to find it in some countries.
PIC18F25520 has some disadvantages:
 big price
 10 bit ADC (12 in 1938)
 no internal Reference Voltage Source for ADC, Vdd is using for this purpose instead.

So, highly recommended to use PIC16F1938 in this design.

IK2HEW, 2020-12-18:
Hi,

thank you so much for your exaustive reply, you fully answered my questions and dispelled all my doubts.

Cheers,

Pino, IK2HEW

DL7BJ, 2021-06-21:
> PIC18F25520 has some disadvantages:
> big price
> 10 bit ADC (12 in 1938)

The PIC16F1938 has also a 10-Bit ADC, not 12-Bit.

https://www.microchip.com/wwwproducts/en/PIC16F1938#additional-features

Dfinitski, 2021-06-23:
Hi, DL7BJ.
Glad to see somebody reads what I wrote here.
PIC16F1938 has 10bit ADC and 3 level reference voltage source - 1024, 2048 and 4096 mV. Each changing the ref voltage in twice adds 1 bit to ADC resolution.
In total - 2 bits in additional or 12 bit resolution.

DL7BJ, 2021-06-26:
Okay, that will do the trick ;-) tnx.