Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #29: all done via pickit2 for 5w. now no display.

07391, 2020-12-17:
All display worked prior to reprograme via pickit.selected for 5w qrp. Programming successfull.Verified.Now no screen.
Any thoughts?
I am new to programming @65yrs so please be kind on any code,Thanks!!

Dfinitski, 2020-12-18:
Hi.
cell 00 to a value of 78, cell 01 to a value of 04, cell 07 to a value of 00, cell 08 to a value of 00 
Read the cell description in the manual about theese cells meanning.

07391, 2020-12-18:
DF.

Thanks for cell info ,will check. After visual think its a solder prob.!! Thanks for reply anyway 73.

07391, 2021-01-07:
Thanks for info! Works fine. My 1st attempt with surface mounts and also use of pickit.Very pleased with result.Thanks again. Tony Batchford.