Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #30: Atu-109

TA3GT, 2020-12-29:
I completed the kit I just bought the problem is when the voltage is released, the contact stays off and one does not release back so I cannot adjust the swr setting everything was done correctly please help

Dfinitski, 2020-12-29:
Hi.
The capacitor's position relay should be under voltage to compensate the inner inductance and get good SWR.
When the tuner board has no power voltage, the normal SWR 1.5 up to 2 on the upper bands. It is a simple behavior.

TA3GT, 2020-12-30:
How can I take a video and send it to you

Dfinitski, 2020-12-30:
Better check with multimeter which relay is always on.

TA3GT, 2020-12-30:
When I energize the system for the first time, will it pull and release the role or it will remain slanted, when it energizes my kit, it pulls to the role and does not let it rotate, even if I try to roost, the roles do not displace I think there is nothing left I can rewind all the coils. sometimes the screen hangs

Dfinitski, 2020-12-30:
I don't understand your question. Did you find what relay stays powered ?

TA3GT, 2020-12-30:
No matter what I do, the error is not fixed, sometimes the screen is stuck. One watt five watt tuner not working. Or the problem may be integrated

Dfinitski, 2020-12-30:
Try 50 Watt

TA3GT, 2020-12-30:
What I'm doing is just winding the coil, assembling the relay and setting up the screen, everything is ready, I don't even have a small margin of error 🤔 (if the board is okay)

TA3GT, 2020-12-30:
My STATİON qrp yeasu ft-817

Dfinitski, 2020-12-30:
You have to rebuild your tuner for reliable QRP work.
See this video https://www.youtube.com/watch?v=dPys_-_wPcQ
The tuner is not able to work with QRP by default.