Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #32: Low Wattage Reading

jimsl04, 2021-01-02:


jimsl04, 2021-01-02:
I completed the kit version.  The unit seems to function properly, but the Wattage readings don't correspond to the Wattage shown on the transmitter or when connected to another meter.  At 10+ Watts input the device shows ~5.6 Watts.  The unit will not function with an input wattage below 9+ Watts.

could I have wound the dual transformer incorrectly?

Dfinitski, 2021-01-02:
Hi. 
Please count how many times a wire goes through core on the tandem-match.
Also, probably installed diodes are not recomended kind.
Check also resistive devider, 34k should be next to processor and 68k should be close to tandem-match.

jimsl04, 2021-01-02:
The primary wires go through 1 time each.  the secondary wires go through ten times each.  Will check the divider asap.

jimsl04, 2021-01-02:
I checked the Resistors.  All correct.  I couldn't tell on thee zeners.  However, While I was checking, it occurred to me that I have ten loops of wire on the outside of the cores, and thus eleven passes through the cores.  Is this enough to cause the problem?

Dfinitski, 2021-01-03:
Yes. Should be 10 passes through the core, not 11.

jimsl04, 2021-01-03:
Got it.  I'll correct and let you know the result.

jimsl04, 2021-01-03:
Windings now correct.  Same problem exists.  Ideas?

Dfinitski, 2021-01-03:
Please attach a foto of your tandem-match.

jimsl04, 2021-01-03:
[image]
[image]

Dfinitski, 2021-01-03:
Looks good.
Probably there are wrong diodes.
I don't have more ideas.

jimsl04, 2021-01-04:
Right.  I'll just live with it (maybe swap out the diodes later).  I really like the device, and will tell others how great it is and how well you support it.  Thanks for your help.

Jim