Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #35: v3.2 source code

rafael2k, 2021-01-15:
Hi,
Here can I find the v3.2 PIC source code? Also, how to burn a now firmware to the pic?

Dfinitski, 2021-01-15:
Hi.
I have uploaded V 3.2 sources.
Use a Pickit 3 programmer.

rafael2k, 2021-01-15:
I'll close the issue. I'm buying a Pickit 3 for me.

rafael2k, 2021-01-15:
Also, sorry the late question - which software should I use to compile and flash the PIC?

closed 2021-01-15 by rafael2k

Dfinitski, 2021-01-16:
You don't need to compile the sources, the ready to flash hex files are here
 https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/tree/master/ATU_100_EXT_board/FirmWare_PIC16F1938
But you can if you want, it is MikroC Pro for PIC.

Programmer's software is here https://drive.google.com/file/d/18wiH5RjL5R8ku7a9MQ4Hxx9SmUZInpvj/view?usp=sharing

rafael2k, 2021-01-16:
I want to play with the firmware. Thanks a lot!