Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #36: Firmware update for Dummies - Is there one?

clsguy, 2021-01-19:
Hi All, I just finished building my atu-100 with version 3.0 and it is working fine. Being an electronics tech really helped.  I would like to update it to ver 3.2.  I can order a pikit3 clone very quickly but I am unsure how to connect it and get started with updating the firmware.
Is there a document out there that would help?
Thanks!

va3rr, 2021-01-19:
The youtube video by IK3SSG should help you out.

[https://www.youtube.com/watch?v=dPys_-_wPcQ](url)

rogerclarkmelbourne, 2021-01-19:
Clicking on that link didnt seem to work for me

https://www.youtube.com/watch?v=dPys_-_wPcQ

clsguy, 2021-01-20:
Thank you for the replies. I reviewed the youtube video which did give me some insight.  I was just wondering if there was something a bit more detailed, like how to backup the current firmware with my 4 line display, how to ensure that backup saved properly etc.  Does the chip have to be erased before updating to the latest firmware? I also noticed there was a voltage setting - does this have to be set for erasing/writing?  
These are the details I was looking for, or am I digging in too deep?
Thanks again.

va3rr, 2021-01-20:
clsguy;
You can use the 'read' button on the Pickit3 app to read the 16F1938 and save the hex file to your laptop.
AFAIK, writing to the 16F138 will write over whatever is there.  I didn't bother to erase mine, but just overwrote it. 

N7DDC de VA3RR.  David, the new 3.2 firmware works great with cheap Aliexpress IIC OLEDs (SSD1306).  I used to get odd characters with the 3.0 firmware, but no more.  Spasibo

Dfinitski, 2021-01-20:
clsguy, you have to start to do something and ask specific questions.
If you would to know how to program PIC controllers in general, you can find a lot info in the Google search:
https://www.google.com/search?q=how+to+program+pic+with+pickit+3&rlz=1C1OKWM_ruUS851US851&oq=how+to+programm+PIC&aqs=chrome.4.69i57j0i13l3j0i13i395l3j0i13i30i395l3.13274j1j7&sourceid=chrome&ie=UTF-8

amsnl, 2022-11-29:
i see the video and the chance to QRP mode and i know to set it in auto tune but hoe to update the firmware to 3.2 i cant find it alsi the files are triki to use some file dont have hex exetion but nothing or zip what say its damage 

please help

Dfinitski, 2022-11-29:
The hex file inside zip file, it's not damaged.

amsnl, 2022-11-30:
i cant unzip it it say empty

amsnl, 2022-11-30:
[image]
like you can see

Dfinitski, 2022-11-30:
Unknown format. Is that Windows ? What kind? Usually Windows opening zip by default.
Anyway, install some free zip archiver first.