Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #38: Strange power indication

diygoodies, 2021-01-24:
Hi I have N7DDC 7x7 tuner and [ SW2013 mini transieveк](http://www.diygoodies.org.ua/?p=1377)and  [windom antenna](http://www.diygoodies.org.ua/?p=1506). Transiever has 14-15W of full power in all bands so I expect from tuner indication of same power value, but all that I get discribed in flowing table.

[image]

I tryed BAT41 diodes with windom antenna, and 1N5711 diodes with windom antenna and then 50Ohm dummy load.
As for now I can't understend why Power indication is so low, and why Power indication is so non liniar when antenna is connected?
Here is a picture of my transformer
[image]
Whaiting for advice. Thanks!

mike-hb, 2021-01-26:
What about reducing the secondary winding down to a few turns -- the sensivity of the power bridge is increasing.

diygoodies, 2021-01-26:
> What about reducing the secondary winding down to a few turns -- the sensivity of the power bridge is increasing.

But what about non liniar mesurments on real antenna?

mike-hb, 2021-01-27:
Non-linear measurement results occur when measuring with low power. Your table shows only small deviations of the SWR between your two measuring devices (columns 1 and 2) only slightly above the minimum power threshold of 5 W -- Could this be the reason? What did you expect when using different rectifiers? As described in the manual, one can find out, that there is a correction of the measured values. (V 3.2 in main.h line 42-60 correction is performed with a step function with 14 steps over the full range) This implies one special type of diodes.

Hope this helps
Michael