Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #39: ATU-100 took itself out of Auto Mode Ver 3.2

sparkey2450, 2021-01-25:
David,
I received my new ATU-100 and it indicated it was programmed with version 3.2. I put it online and it booted in Auto Mode. Did a couple of tunes on 40 and 20 meters at low power. Worked OK. Once I ran a tune I increased the power to 100 watts in CW. The unit at this level tried to do a tune again as I could see the VSWR jump on the Ameritron Swr meter and could hear the relays being manipulated.  Every time I keyed the transmitter it would try and retune at this high power level. It would fine a tune and stop tuning. Then I noticed that the period was gone on the display and the unit had taken itself out of auto tune. I have not experienced this in version 3.0. Since there is only one button on this unit I could see some one getting a tad irritated. I'm going to open up the unit and get a look at the construction and also get a copy of the firmware installed.  Just thought you might want to know. Not a big deal here as I plan on using one of the earlier versions. I believe the unit came from China. 

Steve W8TTY

sparkey2450, 2021-01-25:
David,
I think I know the problem. The company selling these have you order it with or with out the Auto and By-pass buttons. After opening the case I found two nice little antennas connected to those inputs of the processor. I'm going to remove those wires and retest the unit. This units has wires that are too long. Poorly wired. 
Steve W8TTY

Dfinitski, 2021-01-25:
Hello, Steve
Try to power on your chinese ATU with pushed down Tune button.
Most likely you will see a short menu to control main parameters.
Chinese firmware is little bit different then mine.

sparkey2450, 2021-01-26:
David,
Tried the power up with the tune pressed. Went to Fid_loss. No menu. I cleaned up the wiring, removed those two loose wires (Little antennas in side the box) they had attached to the By-Pass and Auto button, put a twist in the DC Power wires. shortened the excessive wire.  Fixed the problem. This can be closed. Thanks for the advice on the power up. They are also using the following PIC16LF1938  for the processor. They had some of that excess wire right on top of the toroid's. You get what you pay for. Thanks again. By the way I figured out where all those conversion numbers came from in conversion(). Takes me a bit longer now days but I eventually get to the right place. Have a great day.
Steve W8TTY

closed 2021-01-26 by sparkey2450

Dfinitski, 2021-01-26:
Hello, Steve
PIC16LF1939 can not be powered by 5 Volts
1.8 - 3.6 is the normal voltage range.