Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #40: Feature request for tuning in the low power range below 1 W

mike-hb, 2021-01-27:
For low-power operation, e.g. for WSPR with less than 1 W, I suggest to introduce a low-power factor -- perhaps in cell 37. 

This way also powers below 1 W can be set for the variable "min_for_start". This allows tuning at the low available power of a QRP-WSPR-TX.

There is already a setting for high power in cell 30. Maybe the setting for low power can also be made there with a changed parameter "-1". - Of course the program and the calculation of the power must be adapted...

Dfinitski, 2021-01-27:
Hello, Michael
I thing with current schematic it is impossible.
You can try to use AD8361 as a linear RF detector.
You can set it up like 1W = 10W in the tuner firmware and use it as is with no changes.

mike-hb, 2021-01-30:
dear David,

thank you for the quick reply.
Your suggestion with the AD8361 is the Gold standard!
But if we look on the momentary layout, I think that by 2 measures yet a rudimentary possibility for tuning is available. Therefore I would like to specify my suggestion.

Currently the problem is that by tuning temporarily the displayed power falls below 1 W and then the tuning process is terminated prematurely.

A good possibility for this would be if the tuning limit could also be set in steps of 0.1 watts. This is currently not possible. On the other side in main.h in the lines 122 to 128 in the function get_pwr the measurement result is adjusted by multiplications and divisions. My suggestion was to influence it in dependence of a memory cell, eg 30 odr 37 these factors to use in this way also powers below 1 W for the adjustment. 

On the other side the transformation ratio has to be adapted. Here one must be careful of course not to let the losses become too large by the measuring.

The shottky diode has a low threshold voltage, but with changed transformer ratio it might be enough... For example: 1 W is 10 V peak on 50 ohm and with transformer 5:1 it is 2 V (the BAT40 has a threshold from 150 mV up) -- this is an error of up to 5 to 10 %. -- This will be sufficient for my purpose. And when we reduce the power, the error will increase.

What do you think about that?

hope this brings us forward to the use at low power

vy73
Michael, DK5HH

Dfinitski, 2021-01-31:
Hello, Michael
You forgot a devider 1/3 after rectifiers in your calculation.
As an experiment, you can set a turn ratio more then actual and  tuner will think that input power is more then actual.
Check if the tuning process works good with 1W input power.

mike-hb, 2021-02-01:
Dear David,

Thanks for the comment.  You are right. The voltage divider was not in my calculation.

But, as I read the circuit diagram, the divider divides the rectified RF signals, which means DC -- so the rectifier has the higher RF voltages from my calculation and by the ohmic 1/3 divider the above mentioned errors of the "knee" threshold voltage will be reduced by factor of 3. -- This supports my idea of the low power tuning range for an experiment.

I checked my tuning of the antenna impedance here -- the atu works perfectly with power just above the 1 W (as mentioned in the discussions in another thread here, the tuner needs twice the minimum power level for correct tuning. My experience is: mostly 1.2 to 1.5 W will does it as well. 

In these experiments I missed the option to decrease the minimum power level down to 0.5 or 0.3 Watt to measure at these levels around and below 1 W.   

vy73
Michael

lu1aax, 2021-07-29:
> dear David,
>
> thank you for the quick reply.
> Your suggestion with the AD8361 is the Gold standard!
> But if we look on the momentary layout, I think that by 2 measures yet a rudimentary possibility for [...]

Did you try setting min power to 0? It worked good for me and the side effect of tuning on no power input is not that bad...