Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #41: Ready Built ATU-100-extended-board reading high swr reading over 14mhz

2E0DMB, 2021-01-28:
Hi Great atu ,and info from you David,

I got a ready made board off ebay for the same price as a kit,
i re-programmed it and set the tandem match to 5 turns as i want to use it with my uk mil radio Clansman PRC320
That puts out up to 30w output but the foldback is to fast,
Now i have set it to work from 1w all is good, 
But after 14mhz The swr reading was going up the higher you go up in frequency if i set 25 ohm it should read near swr of 2:1 but it was reading over 3-5:1 the higher you go,
So i replaced the sensing diodes with BAT41 and all is now correct ,
so i do not know what was fitted but they were not bat41, 
so if anyone is having the same problem, i would change the sensing diodes before doing anything else 
Thanks Dave 2E0DMB

Arm-D, 2021-06-01:
I can confirm the production issue with the diodes. Bought a kit with preassembled SMT components from Aliexpress. The diodes were marked "B66" and they are definitely not suitable for this purpose. 14MHz and up the tuner did not work as expected. After replacing the diodes with BAT41 the problem is gone and tuning up to 50MHz works fine. Also the power measurement gives now reasonable values across the whole frequency range.

Dfinitski, 2021-06-01:
Hi. That is expected.
Also, torroid cores from China have large losses and become hot very fast.

2E0DMB, 2021-06-01:
Thank you for your reply
Great all is now ok
Thanks Dave 2E0DMB

frankliao84, 2021-07-28:
> 
> 
> Hi. That is expected.
> Also, torroid cores from China have large losses and become hot very fast.

you mean core for tandem matching or the inductors. Maybe they are not what are in the BOM of your document list?

Arm-D, 2021-08-02:
The China core for the SWR-bridge seems to be of good quality. I have replaced it with a genuine BN43-202 and there was no difference at all. 
While bulding the kit I have measured the inductance of the matching inductors. All were close to the specified values, so the core material seems to have the correct permeability. Not sure about the losses, I have to doublecheck to see if they heat up more than expected.