Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #43: What is the max SWR this tuner can handle?

psabbag, 2021-02-07:
I couldn't find this information anywhere.

Does someone knows what is the max SWR this tuner can handle?

Thanks,
73,
Pablo, VA3HDL

Dfinitski, 2021-02-07:
Hi, Pablo
I never heard about broken tuner due to high SWR.

Dfinitski, 2021-02-07:
I heard in some conditions the MCU firmware has been spoiled and required reflashing.

psabbag, 2021-02-07:
Hi David,

Maybe my question isn't clear.
I have an MFJ-939 Automatic Tuner that can handle 32:1 SWR mismatch, that is 1600 ohms to 50 ohms, matching impedance.

My questions is what is the maximum matching impedance the ATU-100 7x7 can handle?
10:1 ? (500 ohms)
20:1 ? (1000 ohms)
etc.

Thanks,
73,
Pablo, VA3HDL

va3rr, 2021-02-11:
Pablo - the mismatch any tuner can handle is frequency dependent.  This tuner can only switch in a total of 8uH, so on 160 metres, it's range is quite small.  

Here's a webpage where you can plug frequency, source and load impedances for an L-network, keeping in mind that only the two first network configurations (low pass) are valid for this tuner...

https://leleivre.com/rf_lcmatch.html

psabbag, 2021-02-11:
> Pablo - the mismatch any tuner can handle is frequency dependent. This tuner can only switch in a total of 8uH, so on 160 metres, it's range is quite small.
>
> Here's a webpage where you can plug [...]

Thank you

Dfinitski, 2021-02-11:
Pablo, now I got you. 
What handle - usually used as what handle before be brocken :)
Matching edges for each bands you can find here and also the Python script, you can run it on your PC:
http://www.cqham.ru/forum/showthread.php?38284-%CF%F0%EE%F1%F2%EE%E9-%EC%E0%EB%EE%E3%E0%E1%E0%F0%E8%F2%ED%FB%E9-%E0%E2%F2%EE%F2%FE%ED%E5%F0-%EE%F2-N7DDC&p=1796851&viewfull=1#post1796851

Dfinitski, 2021-02-11:
English conversation 
https://groups.io/g/ATU100/topic/computer_simulation_model_for/80572292?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80572292

psabbag, 2021-02-12:
Excellent, thanks David ! The charts are bigger in the English version. Also I couldn't run the Python script on my computer. For some reason I'm having trouble with the smithplot library:

Traceback (most recent call last):
  File "D:\Download\Radio\ATU-100_7x7.py", line 10, in <module>
    from smithplot import SmithAxes
ModuleNotFoundError: No module named 'smithplot'

I'm not sure what I'm missing. There is no smithplot library to install.

Thanks again
73,
Pablo, VA3HDL

Dfinitski, 2021-02-12:
Hi Pablo
Probably you need https://pypi.org/project/pysmithplot-fork/

psabbag, 2021-02-12:
That worked, but now got a different error:

`freq = 1820000
Traceback (most recent call last):
  File "D:\Download\Radio\ATU\ATU-100_7x7.py", line 53, in <module>
    ax = pp.subplot(1, 1, 1, projection='smith')
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\matplotlib\pyplot.py", line 1142, in subplot
    ax = fig.add_subplot(*args, **kwargs)
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\matplotlib\figure.py", line 1402, in add_subplot
    ax = subplot_class_factory(projection_class)(self, *args, **kwargs)
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\matplotlib\axes\_subplots.py", line 42, in __init__
    self._axes_class.__init__(self, fig, self.figbox, **kwargs)
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\smithplot\smithaxes.py", line 374, in __init__
    Axes.__init__(self, *args, **SmithAxes.update_scParams(instance=self, filter_dict=True, reset=False, **kwargs))
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\matplotlib\axes\_base.py", line 511, in __init__
    self.cla()
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\smithplot\smithaxes.py", line 477, in cla
    self.grid(b=self._get_key("grid.%s.enable" % grid), which=grid)
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\smithplot\smithaxes.py", line 1064, in grid
    for xs in np.linspace(x0, x1, x_div + 1)[1:]:
  File "<__array_function__ internals>", line 5, in linspace
  File "C:\Users\psabb\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\numpy\core\function_base.py", line 120, in linspace
    num = operator.index(num)
TypeError: 'numpy.float64' object cannot be interpreted as an integer`

Dfinitski, 2021-02-12:
A bug in the numpy, you will need *pip install numpy==1.16.6 --upgrade*

actionwatcher, 2021-02-17:
Hi David, 
A related couple of questions. I have a very high SWR on both 3.5 and 7 MHz. On 3.5 the tuner generally can be tuned manually but automatic mode usually does not find the tuning point. On 7 Mhz it tunes most of the times but not all the time. Is it expected behavior for the high SWR loads?
Second question: Do torroids' temperatures depend on the load Z? On 7 Mhz the higher inductance torroids become extremely hot with just one 100 watt CW QSO ~10 minutes. From my brief experiments it can sustain ~ 10-20 watt of CW QSOs in open enclosure. I am not sure about the quality of the torroids (as the kit was from eBay) on the other hand the torroids look a bit small for the 100 watt. The torrid sizes are per your specs and inductance measured prior installation was reasonably close your specs.  Do you recommend replacing torroids with known quality of the same size or increasing the size of the torroids?

Thanks! 73! Arkady/NU6N

va3rr, 2021-02-17:
Arkady - from whom did you buy your ATU-100?  Some sellers are not providing genuine Micrometals T68-2 cores.  Perhaps low Q is the reason for your inductor heating?

[https://owenduffy.net/blog/?p=11873](url)

actionwatcher, 2021-02-17:
That is my suspicion also. I got my kit from ronoe_1231. The seller was quite responsive and stated that kit came from China. The seller is from US. Thank you for the reference!
> Arkady - from whom did you buy your ATU-100? Some sellers are not providing genuine Micrometals T68-2 cores. Perhaps low Q is the reason for your inductor heating?
> 
> [https://owenduffy.net/blog/?p=11873](url)