Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #45: Future Board Layout

sparkey2450, 2021-02-13:
David,
Hope all is well on your end. As you know I've been using your 7x7 Board to write code under MpLabs IDE. I find MpLabs MCC very useful. Anyway to the point. I got where I was configuring the PIC I2C Interface using MpLabs MCC to talk to a Raspberry Pi. MpLabs MCC assumes the PIC code will use pins RC3 and RC4 for the I2C communications instead of the pins you laid out, RB6 and RB7. MpLabs expects the programming tool to be connected at RB6 and RB7. So if you are ever contemplating a board revision might want to keep that in mind. I know I can go in and massage the MCC code to use the other pins. But it would have been nice to keep the debugger hooked up while the code was being developed. I purchased a populated board only so I'll look and see how much trouble it would be to re-rout some traces.  Thanks Dave.
Steve (W8TTY)

Dfinitski, 2021-02-13:
Hello, Steeve
This project uses software I2C, you dont need to care about hardware I2C settings.

sparkey2450, 2021-02-13:
OK. But under the MpLab MCC System when you select  the I2C Library they configure the PIC to use Pin RC3 for the Clock and RC4 for the Data. The compiler you are using is $250. The MpLab System is Free.  If you ever wanted to have this board compatible with the MpLab Compiler and MCC Code Generator would it not be better to have the MpLab System work without having to massage the code it generated. I understand how I2C works.  If in the future you are making board layout changes might be something to consider. That's all I'm saying.

rogerclarkmelbourne, 2021-02-13:
@sparkey2450 

Even better, change to a processor that uses GCC

sparkey2450, 2021-02-13:
David,
If you look at the spec sheet for the 16F1938, they are specifying pins RC3 and RC4 for I2C Communications. RB6 and RB7 for the in circuit debugger.  
Steve (W8TTY)

sparkey2450, 2021-02-13:
Roger, I' started using this because I was interested in putting a Tuner out at my antenna.  David's code works well when you put it n Auto. And you can stick his board out there at the end of the coax. I wanted to do some control from the shack and put this on the WiFi. This the reason I started hacking at this.  The board is very reasonable. I got sticker shock when I looked at the cost for the compiler David was using. This is when I started to port this over to MpLabs Freebee.

Dfinitski, 2021-02-13:
Steve, again, I2C works independetly to compilators.
You dont need to use any compilator's options. Check again the Software_I2C.c

sparkey2450, 2021-02-13:
David,
Your missing the point. The developer of the chips does not want I2C Communication on those pins. It has nothing to do with the protocol. Please read the chip specifications.  I can drive backwards in my car all day long but that's not how the car is designed.

rogerclarkmelbourne, 2021-02-13:
@sparkey2450 

Yes. I know it works well.

I did start designing a board myself using a cheap ARM based processor, and also looking at using something like a ESP32 module instead of the PIC, using KiCAD.

But my other commitments made me shelve the project until later this year.

sparkey2450, 2021-02-13:
Roger,
Well David's Board appealed to my lazy side. My original thought was use a roller inductor and variable capacitor. But then I found this ATU-100. I stripped out all the code that did not pertain to my exact setup. David has lots in there to support many different items. I wanted something simple. Just match the antenna to the coax at the antenna and allow me to see and control from the shack.

rogerclarkmelbourne, 2021-02-13:
Yep. 
I built a Chinese kit and it worked well.

I then bough another Chinese kit with the intention of removing the PIC and connecting another processor via some flying leads

Unfortunately that kit was missing a relay :-(  , and it was not possible to get the supplier, in China to resolve the problem, because they never admit a fault in what they send you.

I also bought a bare 1kW PCB from a Russian vendor on eBay, but it took 4 months to arrive, due to the pandemic, and by the time it arrived, I had too many other projects underway

I've been analysing board layouts from various manufacturers and trying out various layouts in KiCAD 

e.g.

[image]

Looking at improving the SWR when the ATU is not enabled, by reducing the track distance etc

But I have a lot of work with my day job and other ham radio related projects, and don't currently have time to continue with this.

Dfinitski, 2021-02-13:
Steve, my I2C code works with any pins of MCU, it does not use hardware I2C module. There eare some important reasons to do this.
You dont understand what is software I2C, really ? Software SPI ? You should to know what is this if you program MCUs.
Check again the code, no any proprietares calls for I2C are used. All the calls are built-in.

sparkey2450, 2021-02-13:
David,
I'm not going to argue about it. But here's the deal. When you run I2C on pins RB6 and RB7 you are not following the manufactures recommendations. I did not design the processor but the manufacture has indicate where the programmer should configure IC2 communications, which is on RC3 and RC4  They want to reserve pins RB6 and RB7 for the in circuit debugger. Not talking to an OLED.  That's great that it works. love it. But its not standardized. Therefore it causes problems  when trying to use a system like MpLabs which follows the standards. I thing you did a great job. But standards are there for a reason.

sparkey2450, 2021-02-13:
Roger,
I bought one complete box from China. Looked like a 6 year old wired it. Like I told David. I have a bare board and will look to see how much trouble it is to move the I2C Communications. This means I have to rewire two capacitors also to different pins. Then I can use the in circuit debugger. The code I have now has no OLED, Test Software, etc. Its strictly set the power to tune, key the tx, let it tune, when its done, move the power up to operate. The code compiles using MpLabs IDE and XC8.  There are a number of things that need to be changed because the MikroC Compiler uses some functions that are not portable across platforms.  The other thing I can do is just use the MpLabs IC2 configuration and massage it to use pins RB6 and RB7. I'll eventually figure it out. I would much rather be able to connect the PicKit 3 and use the debugger on the PIC and then watch the raspberry pi code. I was going to stick with MickroC but when I finally figured out how to use MpLabs MCC generator, it really makes the configuration a piece of cake. You can concentrate on writing the application instead of trying to figure out how to setup the chip. MCC write everything you need to work with the processor without all the head aches of configuration.

rogerclarkmelbourne, 2021-02-13:
@sparkey2450 

I built mine from the kit, very carefully ;-)....

I understand what you mean about the pins that David chose to use for I2C (which from what I gather is software I2C).
I presume David chose to use those pins to make the PCB layout easier, but prevents you using the in-circuit debugger.

I mainly use ARM based MCU's and only have an older PicKit2, which is not compatible with this PIC :-( :-(

I'm loathed to buy a PicKit3, just for this, and my intention was to remove the Pic from the 2nd kit I bought and probably initially port the code to something like a STM32F103 aka "BluePill" board.

BTW.

I had a look at David's tuning code, but I didn't have time to understand the algorithm, or even if there is a full algorithm involved.

I found another ATU project which is very similar, (Sorry I can't find the link at the moment), and the guy went into great detail about an efficient algorithm based on a 2 dimensional matrix of readings, which iteratively reduced the size of the L and C it was switching in and out, to get to the optimal tune point very quickly.

However, although the guy posted some information, he chose not to post his code.

I understand the guy's decision, as I get quite jaded by spending a huge amount of time doing research and then writing optimised programming, only to find that someone takes the code and uses it in their commercial or open source projects in breach of the copyright conditions on my code etc.

I'm sure its been a double edged sword for David with the Chinese banging out these kits, as it leads to a load of dumb-ass support questions in this repo, rather than anyone actually contributing.

Dfinitski, 2021-02-14:
Steve, RC3 and RC4 are reserved to use **by hardware I2C module** because it is restriction, not a standard. Hardware modules can not physically use any MCU pins. Each hardware module has own dedicated pins which working with it.
But I did not use MSSP module in my design, because:
1) No free pins to use them only as I2C bus 
2) OLED display is an option, normally a tuner should to work close to antenna in automatic mode.
Initially, LCD 2*40 was added to help  debug the code, to show some register info.
3) Software I2C realisation allows to work safely with errors on the bus and also with/without LCD/OLED, wrong address device, collisions, RF interferences.
Hardware realisation with MSSP module requires much more details to reach this, interrupt control and timer count using.
Bad code could to tight processor easy.
My code is not using interrupts and timers, except WDT timer.

> I found another ATU project which is very similar, (Sorry I can't find the link at the moment), and the guy went into great detail about an efficient algorithm based on a 2 dimensional matrix of [...]

Probably I have tried all existing cases in the algorithm. Some cases work really fast but only with some kinds of loads - when L or C is 0 an example, but the real problem to find an algorithm wich works good with all the kind of loads. Unfortunatelly those ones are not to fast as I wanted. I was forced to iterate over and flexibly configure some constants because simple mathematic model does not work as expecpted.

sparkey2450, 2021-02-14:
Roger,
I did not pay too much for the Pickit3. I got it on Amazon.  I started out looking at this from the automatic stand point but as I got into this I started to think why automatic? The purpose is to match my antenna to my setup. So although I've got the automatic portion working once I get this to interface to something in the shack and can control the ATU from there. Then I will just setup a block of memory where you put in the L and C Values you need to match the antenna on that band and select them as you need them. Some times the simplest answer is the correct one. I have an MFJ 939 that puts the tuning values into memory. In their case they also read the frequency . So once you key the TX it goes back and sees if there is a match for this frequency and then just plops in the values needed.  I'm looking at this more as a replacement for a manual tuner that you control from the shack.

rogerclarkmelbourne, 2021-02-14:
@sparkey2450 

Australia has Amazon in name only. 
They carry virtually no local stock. I don't normally waste my time looking on Amazon.com.au 
However, I just had a look, and there were only 3 listed, starting at $28 inc shipping, all from China, the most expensive was $255, but had free shipping ;-)   Standard delivery time was in April, fastest was the end of March.

There are some vendors on eBay, who claim to have local stock, but often this is BS, and they charge full price and the just try to drop ship from China, which never turns out well

So I don't think I'll bother.

sparkey2450, 2021-02-14:
Ok Roger. Nice chatting.

closed 2021-02-14 by sparkey2450

rogerclarkmelbourne, 2021-02-14:
@sparkey2450 

No worries.

sbridger, 2021-11-15:
@rogerclarkmelbourne 
    trying out various layouts of my own in KiCAD

Could you share your Kicad?
I need to make a specific PCB form factor, and the schematic and footprints would save me a bit of time.
thx

Dfinitski, 2021-11-16:
I did not use Kicad. But you can try to export to Kicad from Easy EDA.

sbridger, 2021-11-16:
> I did not use Kicad. But you can try to export to Kicad from Easy EDA.
I meant Roger's kicad version...

So the ,lay6 file is easy eda, with the schematic embedded?

No luck trying with https://wokwi.com/tools/easyeda2kicad
or https://easyeda.com/editor

Dfinitski, 2021-11-16:
No. Lay6 is a Sprint Layout 6 file.