Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #46: Problema

Celta1960, 2021-02-18:
Hola amigos, tengo un problema el cual no se a que es debido en ver 3.0.Funcionar funciona a la perfección pero en ciertas frecuencias y todas las bandas me da una señal fuerte hasta el punto de no poder utilizar en esos Mhz.
     Adjunto fotos con el puesto y quitado. A que es debido?
     Un saludo
[image]
[image]

Dfinitski, 2021-02-18:
Hello, Celta1960

New in the version 3.2
Fixed bug with continuous data transfer to the display.

So use 3.2 version of firmware.

To check this out, disconnect the display and try without it.

Celta1960, 2021-02-18:
Hola Dfinitski, probare a instalar la versión 3.2 y comentare si se corrigió el pequeño fallo.
   Un saludo

Celta1960, 2021-02-18:
Gracias amigo probare la 3.2.Un saludo

Celta1960, 2021-02-18:
Hola de nuevo, con la pantalla apagada o desconectada sigue igual, sera cosa de cambiar el firmware?.
Saludos
[image]

Dfinitski, 2021-02-18:
I think that is bad power supply for the tuner board. Try to feed it with
batteries, temporary for the test.

Celta1960, 2021-02-18:
No creo que radique el problema en la alimentación ,ya que solo es en determinadas frecuencias fijas en las demás perfecto, no optante probare para salir de la duda.
   Saludos