Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #48: OLED Display inverted

PA3FUN, 2021-03-05:
Hi David,
Help :).
I just updated the firmware of my ATU to v3.2 using the appropriate Github Hexfile to find out the OLED 128x64 display is now inverted. My question: would it be possible, using a hex-editor, to revert the display back to normal?
I checked out the source-code of "oled_control.c" and found line 186: Soft_I2C_Write (0xA6); // 0xA6 - normal, 0xA7 - inverse). Would this do the trick? And if so, would hex-editor suffice to change this single item?
What string would I than need to search for? I looked for A6 but got stranded..
Thanks for your help and for this inspiring project!
73's Christiaan Roselaar PA3FUN
[image]

Dfinitski, 2021-03-05:
Hi.
Read the manual. Cells 03 - 05.

PA3FUN, 2021-03-05:
Hi Dave,
Thanks for your prompt reply!
Am I reading the wrong manual/version? (https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/blob/master/ATU_100_EXT_board/ATU-100_Extended_Board_User_Manual_eng.pdf).
Cells 03-05 don't seem related to display-matters?

PA3FUN, 2021-03-05:
Oops..
I accidentally closed this issue, my mistake, sorry.
Still looking forward to your reply to my previous post :)

PA3FUN, 2021-03-05:
Ah. Values #03 and #05 in cell #01 are for inverted displays!
Dave,
All clear now.
And you were right; I should have read the manual better, it's in there all right.
Sorry to have wasted your time.
73's Christiaan PA3FUN

closed 2021-03-05 by PA3FUN