Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #50: After tune, during TX ATU auto switch to Bypass mode and back to normal

scalpya, 2021-03-25:
Hi
I biuld Atu 100 7x7 external board few days ago. Its so fast and small atu. I discover some problem. When i tune on any frequecy, and i try make some qso and start talking atu after few secunds swtch to Bypass mode and antena not in tune, SWR rise, after another few secunds back to normal and working like this in loop. I no idea where is some problem. 

Thanks

Dfinitski, 2021-03-25:
Hi.
The most likely this is RF interferences. Reduce a wire length to OLED, buttons to minimum. Install 0.1 uF caps to additional buttons pins and ground.

scalpya, 2021-03-27:
> Hi.
> The most likely this is RF interferences. Reduce a wire length to OLED, buttons to minimum. Install 0.1 uF caps to additional buttons pins and ground.

Thanks, Working now.

closed 2021-03-27 by scalpya