Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #51: Power Accuracy

ewagnerjr2000, 2021-03-29:
Hello, i have updated to 3.2 and i have noticed some power accuracy issues. For example 20 watts on my IC-7100 shows 16 watts,  50 watts shows 40 watts on the atu-100 and 100 watts shows 65 watts on the atu.  Just seeing if this is normal?  

Thanks,
Ed

Dfinitski, 2021-03-30:
Hi.
No, that is not normal and changing a version does not affect power
measurement accuracy.
Except the case if you have changed cell 31.
So first of all check if the actual turn ratio of the tandem match is
matched to cell 31.

zelea2, 2021-04-09:
I don't get how to calculate the max power for a given transformer ratio.
The PIC max range is 4.096V. The resistor divider is 1:3 and this brings the range to  12.288V. The rectifing diode is an extra 0.4V to 12.7V. The tandem match ratio reduces the line voltage N times so for 1:10 we have 127V in the line. Power in a 50ohm impedance is P=U^2/Z = 322.6W. For QRP 1:5 tandem match ratio P=80.6W. Is there an extra divide by 2 because only half of the sine wave is averaged?

Dfinitski, 2021-04-09:
Hi. Read here

https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/17

zelea2, 2021-04-09:
I forgot to divide by sqrt(2). The measured value by the PIC is the peak value not the V(RMS). Thanks for the pointer.

w0ati, 2021-08-07:
Can I ask what code revision you were running before updating to 3.2? I have also had the accuracy problem you mentioned.