Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #52: Massive Interference

HB9HIL, 2021-04-05:
Hello Guys
I have the problem my ATU-100 does massive Interference on my radio. I tested with a dummy load. Between S5 and S9+ is way too much. Where could be the problem? I’ve bought this one assembled from Amazon. Disconnected the internal battery, display and charging regulator. Nothing changed... Some experience with it? Saw the same interference on a YT video on 6 meters but it’s damn hard to find a solution. It seems to be the china version and maybe it’s the IC (PIC16F1938). But before I start soldering, maybe it's a good idea to ask here.

HB9HIL, 2021-04-05:
Maybe I should say as well. I just updated to 3.2. Nothing changed.

Dfinitski, 2021-04-06:
Hello.
Strong interferences even without battery (power on the board) ? Check connecting cables, probably no ground connection.

HB9HIL, 2021-04-06:
Sorry. For sure with power. Butbit doesn't matter if I use a Car Battery, a power supply or the internal battery.

Dfinitski, 2021-04-07:
I can only guess you have chinese 5 Volts version of the tuner with step UP/Down power converter.
Check the relays, are they 5 volt versions ?