Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #53: Different Power Reading On Display

PartyRock916, 2021-04-17:
When I run 100 watts of transmit power and press the Push To Talk button on my Yaesu FT-891 radio, the ATU-100 says the following on the display:

PWR = 50W
SWR = 1.19
ANT = 36W
EFF = 71%

I also get the following readings after I release the Push To Talk button:

L = 0.45 uH
C = 100 pF

Question: How come I'm not running the full 100 watts and only 50 watts as seen on the display? How do I increase my EFF % as well?

Thank you in advance for your help.

pimol, 2021-04-17:
Which mode? Only FM or digital arrives to full use of power, actual power depend from voice mean value...

Pietro I2OIM

PartyRock916, 2021-04-17:
> Which mode? Only FM or digital arrives to full use of power, actual power depend from voice mean value...
> 
> Pietro I2OIM

SSB. On 20 Meters.

pimol, 2021-04-17:
Atu computes a few db for the line losses, I dont remember how many. But SSB can't go over 60/70% of the power for a linear amplifier or transmitter. So don't worry about how many they are, try to have te max you can.
Try with FM mode to see eff power, if your TX is able to. Or with tune key if any.
Pietro

PartyRock916, 2021-04-17:
> But SSB can't go over 60/70% of the power for a linear amplifier or transmitter. So don't worry about how many they are, try to have te max you can.

Just to confirm, I will never be running near close to my maximum of 100 watts using the ATU-100 in SSB?

PartyRock916, 2021-04-21:
?

Dfinitski, 2021-04-21:
Party rock, did you check the power level in FM mode ? What is result ?

PartyRock916, 2021-04-23:
> Party rock, did you check the power level in FM mode ? What is result ?

The power level in FM mode is still the same, 50W.

Dfinitski, 2021-04-23:
Try to change a cell value, I don't see other way.

PartyRock916, 2021-04-24:
> Try to change a cell value, I don't see other way.
> […](#)

I apologize but I'm very new to all of this, but can you please elaborate on how I go about changing a cell value and which cell? Thank you.

Dfinitski, 2021-04-24:
cell 31, read the manual, at the end.