Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #55: 0 ohms between RF center pin and ground?

edwardstorlien, 2021-05-03:
Hello, I am currently assembling the ATU-100 antenna tuner, but I noticed by coincidence something I think is wrong.

With a multimeter I noticed my ATU-100 board has approximately 0 ohms resistance between the RF center pin and ground. This is applies to both the RF input and RF output.

I can see on the schematic that the RF input's center pin is connected to RF output's ground through T1, but is it really 0 ohm resistance? Can anyone confirm that this is correct? I am unsure whether I have made a bad solder joint somewhere...

Thanks!

Dfinitski, 2021-05-03:
Hi
Tandem match coiles are connected between center wire and ground for input
and output connector.

edwardstorlien, 2021-05-03:
Hi again, thanks for answer.
Does that mean my multimeter will read 0 ohms, right? Nothing is wrong with my board?

Dfinitski, 2021-05-03:
Yes, that is correct. There is a short by DC, and Hi Z by RF.

MOUSE1025, 2021-05-03:
Dave,
You have the patience of a saint! Excellent support.
Kind regards,
Christopher Paul Ryan

edwardstorlien, 2021-05-03:
Thank you so much for answer, Dave. Great support, I really appreciate it!

closed 2021-05-03 by edwardstorlien