Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #56: Icom/Yaesu support

ast, 2021-05-05:
I don't know if there's any interest in it, but it seems to me it wouldn't be terribly difficult to add Icom (and possibly) support. The ATU-100 would probably work very well with the IC-705.

I'd take a look myself but I don't have access to the mikroC toolchain.

Icom tuner interface:
https://hamoperator.com/HF/AH-4_Design_and_Operation.pdf

Yaesu tuner interface:
https://gist.github.com/ast/cc62dff65d35c857c3a2bd8dfb40051f

73 Albin SM6WJM

Dfinitski, 2021-05-06:
Hello, Albin.
Thanks for the question.
The Icom interface connection has been added in the first firmware of this tuner.
You can read about RA6 and RA7 ports in the manual, "Using the base model".
The main button port is using as START line and RA6 (inverse TX request) is using as the KEY line.
Optionally you can use RA7 noninvert output together with 2N7002 transistor to make an open collector cirquit if it is required.
That time I did not have any Icom rig so I personally did not check this feature but other users told me about successfull connection.
I was focused to full automatic mode for this tuner because it allows to use the tuner with any rigs and also remotelly with no any connections for control.
I am ready to fix bugs if you found them .
Now I  have no any Yaesu rig so I can't work on implementation of the connection. Technically it is possible to use RA6 and RA7 ports as UART RX and TX data lines for data exchange with a rig.

ast, 2021-05-06:
Hi David!

Thanks for your great work!  And thanks for the input. I think that should work, I will try with my Ic-705. If you look at page 7 of the pdf I link to you can see that  failed tuning is signaled by asserting the key for a short period after. Is that also implemented?

Dfinitski, 2021-05-06:
Hi
That is a tuner reset signal, as you may know, the Tune button works also as a Reset by short pressing.

Dfinitski, 2021-05-06:
Hi
No, the tuner does not have that signal inside, it does not know what is this a tune fail.

soosp, 2021-12-27:
Hi,
I prepared my ATU-100 to work with my Icom IC-706MKIIG transciever. An adapter card can be found on Aliexpess and another sites has been used for interfacing with the radio. The schematic can be found at https://ae01.alicdn.com/kf/Hf7f5efcb98bc49de88d1762b71441d97v.jpg
If you press the tune button on ATU-100 the /RA6 line initiates the tuning process on the radio. It switches to CW mode and starts transmit the carrier at 10W for max 20 sec.
If you want to use the tune button of the radio you have to make a connection between the tune button and the START line of the radio with and 1kohm resistor. It is missing from the schematic. By default the START line pulled up to 5V via an 10kohm resistor of this circuit. This reports to the radio that there is a tuner connected. If you press the tune button the radio pulls down this line to GND, but without the connection the tuner doesn't start the tuning process.
The optional Icom switch on the schematic enables/disables this Icom support.
Disclaimer: It works for me but there isn't any guarantee that it will works with another radio.

Hobby4life, 2022-05-06:
Can you share some pictures of the adapter pcb installed?

soosp, 2022-05-06:
Hi,
Here is the link of the PCB:
https://www.aliexpress.com/item/1005001949768630.html?spm=a2g0o.order_list.0.0.6e071802HgZSXf

soosp, 2022-05-06:
Here are some photos about my solution:
[image]

[image]

[image]

soosp, 2022-05-06:
The necessary additional resistor is here:
[image]

This is the modified schematic:
[image]

soosp, 2022-05-06:
My ATU is powered permanently from the radio via Icom tuner interface, and switched on/off by the power switch of the radio, because the radio needs to be an ATU connected at power on to detect it. The power switch works when use my ATU as traditional automatic tuner (Icom interface disconnected).

soosp, 2022-05-06:
[image]