Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #57: rx problem

serdartncl, 2021-05-06:
hi. firstly sorry for my english. i have a problem for my atu100. It was working well before but now when i on rx i can hear just noise but its like static noise. but tuner is works on tx and sending rf. i use with my qrp device under 2 watts. atu100 not modified for under 5 watt. if i tx up to 5 watt may be this problem solve or can i have to do anything?

Dfinitski, 2021-05-06:
Hi.
How it is possible to use ATU-100 with no modification ? The original board does not start a tuning process with power below 5 Watt.

serdartncl, 2021-05-06:
i dont need to process now. i just using for power and swr reading. i using for mono band and dipole for this

Dfinitski, 2021-05-06:
Ok, I got you.
I don't know what could happened, but most likely you are hearing interferences from the display wires, if the FW fersion below 3.2.
Also, if your tuner uses power supply from DC/DC converter , it can be a source of interference too.

serdartncl, 2021-05-06:
bir kaç gündür böyle bir problem yoktu. step up dc regülatör kullanıyorum farklı bir adaptör ile deneyip tekrardan bilgilendireceğim. teşekkür ederim

serdartncl, 2021-05-06:
There was no such problem for a few days. I am using a step up dc regulator, I will try it with a different adapter and inform it again. thank you

serdartncl, 2021-05-06:
I guess, as you said, the problem is in the regulator. I've never thought of it before, thanks again :)

pimol, 2021-05-07:
There are difference if the ATU power is on or off? It seems a problem with
some connector...

I2OIM

serdartncl, 2021-05-07:
Yes ita different. When atu100 is off i didnt hear static noise. When i turn on its coming