Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #58: part parameters

KD4IYI, 2021-05-18:
I ordered and received a couple of boards that were supposed to have all SMD/SMT parts premounted on them, the first had 3 of the 1uf capacitors shattered, and the second didn't have C1-C7 in place, and the ones sent are physically far too small for the pad size. I need to order correct replacements. so, my questions are 
A. the 17 EA 1uF capacitors, the ones for all the relays for instance, what part number, package type, and or type and specification are they? 50v? 1kV? tolerance?
B. the C1-C7 capacitors, what package type are they, I found a listing on the schematic of silver mica 500v, but no tolerance, and no package sizes
These questions I have are for future reference... I saw another issue stating that these aren't frequency dependent, but they don't operate very well at 50MHZ due to internal capacitances? can you elaborate a bit on what would be needed to add that band or modify a tuner to cover it and perhaps sacrifice some lower frequencies? what would be necessary to build a tuner up to 200-250W using this basic layout. I know there are some kits out of Russia and the Ukraine up around 1KW, but I'm not where I can afford those yet.

I also wanted to tell you I appreciate all the work that you have put into this project!
Thanks!

Seannon, AG0NY (formerly KD4IYI)

Dfinitski, 2021-05-19:
Hello, Seannon
This tuner does not work properly on 50 MHz. Anyway, you have to have normal resonant antenna on that band, simple long wire with length 3 and more lambda will not work even with SWR=1. 
1uF capacitors like this CL21B105KAFNNNE
C1-C7 like this CD15CD100DO3
Also you can use ceramic capacitors like this https://www.digikey.com/en/products/filter/ceramic-capacitors/60?s=N4IgjCBcpgLFoDGUBmBDANgZwKYBoQB7KAbRADYAOAZkvgIE4mBWCAuABjngF0CAHAC5QQAZUEAnAJYA7AOYgAvu3IIQySOmz4ipEAHZ95ahxB8QQkeOnylBAEwdYDNRq24CxSGTAcABPwAYiAO9gHBBLD64SHgHP5BsfaOMeaWkCAAqjJSggDyKACyOGhYAK4SOHYgALT2rlCSZTpeZMxmisog9d7gYP1%2BiGj8aIi5hBIdQA

KD4IYI, 2021-05-19:
Thank you very much!, now to order capacitors and get to building!

closed 2021-05-19 by KD4IYI