Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #59: Tuning at low power fails for severe mismatch due to reduced TX-power

JensZander, 2021-06-07:
Hi David,

I recently bought the ATU100 Extended (China mfg, firmware 3.2)  and modified to tuner for QRP according to the nice video by IK3SSG. (5 turns in tandem-match + parameter updates in EEPROM - min power cell 05  = 01
No problems everything looks OK ..maybe the power reading is 30-40% too low, but I can live with that,

I am using the tuner with my ICOM-705 (5 - 10W) .  When I use the tuner to match "not too bad" antennas   with SWR <1:4 or 5, there is no problem, the tuner works excellently and quickly finds a good LC-setting.

However,  when facing a bad antenna (say a random wire),  the tuning process sometimes seems to present the TRX with a so bad match (typically a large shunt-capacitance) that the (measured) input power drops below 1W and the tuning process stops.  This seems to happen intermittently .. sometimes I seem to be  "lucky" and the tuner tunes properly, but most of the time it stops (due to dropping below the input power threshhold?) . When connecting the TRX  to the same antenna directly,  the SWR is horribly high (say 1:7 or 1:8), but the TRX power stays up - so does it if I do reset the LC-setting to zero - so it must be the tuner that makes the match worse)

Is this a known problem or is this a faulty device ?  Unfortunately, the device has a single "Tune"-button".  I cannot see that the RB1 and RB2 pins are readily available to force the device into test modes. 

Can I calibrate power meter do give a correct reading? I think that migth solve the problem. The measure input power when the tuner stops is typically 1.5W (when the tuner indicates 0.9W) 

Jens / SM0HEV

Dfinitski, 2021-06-07:
Hello, Jens
Most likely bad RF diodes are used in tuners from China . Change them according to BOM list .
This tuner has pretty good accuracy around 10% .
Resolve the problem with power measurement and the tuner will work better .

JensZander, 2021-06-07:
Thanks David,
I will check the diodes. I guess you can replace the surface mounts with standard package diodes (1N5711)

Question .. in the tuning algorithm  .. how is the filter configuration determined, i.e. the setting of S8 (capacitance of the TX or antenna side). I am operating portably mainly with electrically "long" non-resonant end-feds with high impedance (i.e. capacitance on the antenna side), I guess the problem mainly occurs when the algorithm tries the low-impedance settings.
/jens

Dfinitski, 2021-06-08:
Hi, Jens
The algorithm remember last successful tuning position of switch and uses
it for next tuning process

JensZander, 2021-06-09:
Hi David,
Mea culpa .. I misinterpreted the number of turns on the tandem match.. It looked like 5 turns - but the wire actually passed 6 times through the core. Now the power reading is correct within 10% -- at least ..up to 17m then the reading starts droping to show 50% of the measured value on 10m  .. that could  be du to the "chinese Schottkys" ?  Tuning of my  random wire works fine .. now up to 17m .. then the power drops below 1W sometimes during the tuning process and the process stops.  Will replace the diodes as well.

Dfinitski, 2021-06-10:
Hi, Jens.
Chinese zenners can not to work as a shottky.