Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #60: AT-100 EXT does not work with TYT TH-9800 band 29 and 50 Mhz

SergioIU2, 2021-06-11:
Hello while with a QRP on 7 Mhz, exactly the famous BITX 40, it works wonders, if I connect it to the TH-9800 on the 29 or 50 Mhz bands it doesn't work. Tune when you want and to make it work I have to increase the minimum power to 40W, while with the BITX 5W they are enough and advance. Then I noticed that both off and in bypass the 70cm band does not pass in rx, there is a considerable attenuation of the signal in rx both in 70cm and in 2mt
Do I have to do something to be able to make it work on the TH-9800 on the 29 and 50 Mhz bands?
And to make the 2m and 70cm bands pass without attenuation?
NB I use a 430-144-50-29 Mhz quad-band antenna
Thanks and sorry for my English!

Dfinitski, 2021-06-11:
Hello, Sergio
Most likely 9800 drops out power level when SWR more than 3.
This tuner board can not to work with 6m, 2m, 430Mhz bands even in bypass mode. Too much losses.

SergioIU2, 2021-06-12:
> Hello, Sergio
> Most likely 9800 drops out power level when SWR more than 3.
> This tuner board can not to work with 6m, 2m, 430Mhz bands even in bypass mode. Too much losses.

Hello and thanks for the reply
That does not work on 144 and 430 Mhz I knew this! And I already have a change in mind to completely bypass everything! But on 50 Mhz it must work! In the photo it shows from 1.8 to 55 Mhz, think that I bought it just for 6 meters

https://it.aliexpress.com/item/4000773678364.html?spm=a2g0s.9042311.0.0.7f914c4dca0OuR

Dfinitski, 2021-06-12:
Hi, Sergio.
No, for 6m band you will need to use a special kind tuner, not like this.
But better to use resonant antennas without any tuner.

Arm-D, 2021-09-11:
Sergio,
please check the diodes. In some tuners the wrong type of diodes are fitted. 
After changing to genuine BAT41 the tuner works up to 50 MHz and needs 5W or less to tune.

SergioIU2, 2021-09-11:
> Sergio,
> controlla i diodi. In alcuni sintonizzatori sono montati diodi di tipo sbagliato.
> Dopo essere passato al BAT41 originale, il sintonizzatore funziona fino a 50 MHz e ha bisogno di 5 W o meno per sintonizzarsi.

Grazie! Controllerò!