Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #61: Power and SWR reading is wrong

w0ati, 2021-06-16:
I am stumped and hoping for advice.
I built an extended ATU-100 kit. Placed a dummy load on the ATU output and applied a QRP transmitter to the input. The dummy load was tested with a NanoVNA and has an SWR of 1:1 at 14MHz. The QRP transmitter (14MHz) output power has been measured with a Diamond SX-600 SWR/Power meter. Its 2.5W. However, the ATU in bypass mode says the power is 1.9W and the SWR is 1.8:1. 
I did the same with an IC-735 at 10W just incase its a power issue. Same kind of results, SWR and power are significantly off.
I put the NanoNVA on the ATU input and the SWR of the dummy load through the ATU in bypass mode is 1:1.
I took T1/T2 off the board. Applied a DC supplies to the 50ohm resistors at the RVS and FWD sense points. The diode Vf is reasonable (1V). The voltage division from the diode output to the MCU input pins is correct. And the MPU measures the voltages and responds to the input voltages with a power and SWR measurement. With some arbitrary FWD voltage and no RVS voltage SWR displayed is 1:1. The board voltage sensing circuit seems to be working.
I re-wound T1/T2 (10:1 ratio). Then tested it on a bread board I made (the transformer, 14MHz source, 50ohm load and a dual channel scope). With 50ohm load FWD gets a sinewave and RVS is 300 time lower. With an open or short for a load, FWD and RVS are equal within a few percent. The transformer seems to be working.
However when I soldered it back into the ATU I still get incorrect power and SWR (no change from before).
I have considered that I have the phasing (winding direction) of the T1/T2 winding incorrect, but I have studied pictures of units I find on the web and can't see that I am doing anything different than I see in the pictures.
As I say I am stumped. 
Thanks for reading this.

Dfinitski, 2021-06-16:
Hi.
What about 1.8 MHz band ?

w0ati, 2021-06-16:
Have not tried it there. Only 14 and 21Mhz. I did sweep the transformer from 1 to 30Mhz looking for changes. There were some but nothing that concerned me. Thanks

Dfinitski, 2021-06-16:
Transformer is good, take a look at the diodes, if you buy the kit in China
most likely the diodes can't work on high frequency.
You can also check a voltage after diodes and correlate it to SWR on the
display.

w0ati, 2021-06-16:
I did measure the sense voltages. With my QRP transmitter (2.5W) Vfwd is 1.01V and Vrvs is 0.29. That is an SWR of 1.8 which is what the display showed at the time. Confirming the board circuity and MCU is working just fine. The problem is, with the dummy load (which is a perfect load at 14MHz) and if the transformer is working, where did that Vrvs of 0.29V come from? There is something going on around the sense circuit (transformer, 50ohm sense loads diodes, capacitors).
I did get the kit from China. I guess I should look into those diodes. And verify the capacitors, I guess. Hate working with SMT components now. I miss the microscope, soldering station and lab stock I had before I retired.
Thanks

w0ati, 2021-06-19:
You are right. It was the diodes. They were are functioning diodes, they just have too much junction capactiance.

k9nud, 2021-07-04:
I just completed my Hisonauto kit from AliExpress and while it works, I'm seeing SWR issues. My KX3 (with ATU bypassed) shows a different SWR than the ATU-100, which always indicates a higher value. I'll assume the RF diodes are the issue. I saw somewhere that in addition to the BAT41 SMDs, 1N5711 traditional diodes are also specified. This board has pads for traditional components too, so shall I assume they'll work? Also, it would not tune my EFHW on 160M, though the KX3 will. Would the diodes be a culprit?

On another note, this model (Hisonauto) has firmware 3.2 and claims to work with 1 W. It still specified the 1:10 tandem match. It seems to tune at 5 W. Any thoughts on whether I'll need to rewire the transformer and reprogram the PIC to get it working with lower-output radios?

w0ati, 2021-07-04:
My SWR, with the diodes that came with a kit, also read high (very high). So following the advice given on this forum I grabbed some thru hole Schottky diodes I had in my part stock. I did not know the generic part number or specifications, however, given where I got them, I expected them to be high frequency. The pads on the PCB are really two sets, one SMT size the other larger. So I simply surface mounted the thru hole leads on the larger pads. The SWR readings then because more believable. I never checked the accuracy but the reading were much better.I then looked for the MAT41 and 1N5711 diodes on the internet. They both came up as thru hole. Rather than enter a drawn out search for SMT parts, I bought the MAT41 from Digikey and surface mounted the thru hold leads as before.I have not completed all the testing I want to do yet. I have only tested the tuner on a random length dipole I have in the attic for SWL. I used the ATU test mode to manually tune the antenna on the 20 through 10meter bands using a NanoVNA as the SWR meter. I noted the turner configuration, L, C and SWR for each band. I then used 9W from my IC735 and let the ATU tune that antenna. It did alright. In all but one band the tuner came up with the same configuration and LC values I did. The one case where it did not, I think the algorithm just has a threshold where it says that is good enough and stops. The difference in SWR as per the NanoVNA was 1.02 vs 1.06.At 9W, I don't think the ATU power reading is accurate. It says 7.8W were the Diamond meter says 9W. This could be a diode linearity thing and get better as the power increases (have not tested that yet)The SWR also does not agree with the NanoVNA or Diamond SWR meter although its not bad.1.1 vs 1.08, 1.05 vs 1.09, 1.15 vs 1.18. What is important is that the minimum is real. And it is.
I did not test the tuner at any frequency lower then 14MHz. That is because the SWR/impedance of the dipole so far off I don't think the tuner is going to be able to do it.
And I have not tested it on my QRP (2.5W) radio (yet).

k9nud, 2021-07-04:
Thank you. I ordered ten from Mouser. The cost for ten is the same as for six. I will report back once I replace them.

I'm thinking I might build a multi-band vertical and put one of these at the base. Not sure if I'll power it with a Bias-T or a small solar panel and some 18650's.

lu1aax, 2021-07-21:
Hi, I assembled china kit with 3.0 fw

I put it in bypass mode and connect a dummy load (rated up to 6 GHz) at output

In the 80M band I get

PWR: 10W
SWR: 1.01
ANT: 8.3W
EFF: 78%

Since I'm in bypass mode and a dummy load with a virtually perfect match I'm puzzled about why I am seeing 8.3W getting out when 10W are getting in.  SWR reading looks good though.

In the 10M band I get

PWR: 10W
SWR: 1.15
ANT: 8.1W <-> 8.5W
EFF: 77%

Here, SWR reading seems high for this load. Power output is not that stable. May be this is somehow related to the diode problem mentioned in previous posts in this thread?

w0ati, 2021-07-21:
I have personally concluded the power bridge is not accurate. Thus, the power reading is not accurate. However, SWR is a ratio of two equally inaccurate power readings so it ends up correct.

Dfinitski, 2021-07-21:
Hi.
The 3.0 firmware uses 1.3db fiber loss by default.
Renew it to latest, that uses 0 dB by default and a possibility to change the value any time.

w0ati, 2021-07-21:
Mine came with FW Ver 3.2.

Dfinitski, 2021-07-21:
That's good, so you don't have that problem as lu1aax.
Did you change Chinese diodes in the tandem match measuring circuit ? That is famous problem.

w0ati, 2021-07-22:
Yes, I replaced the diodes. The ATU will do a reasonable job of detecting a SWR null and matching an antenna. However, I don't trust any of the power readouts on the display.

Dfinitski, 2021-07-22:
That is weird, the power measurement accuracy is pretty good on this device, even better than some gauges type manufactured meters .
Check the resistors in devider after diodes, 68k and 34k close to processor.
Try to change tandem match ratio.

w0ati, 2021-07-22:
I agree, its weird.
Here are my power readings using a dummy load and a Diamond SWR/Power meter as reference.Diamond reading    ATU reading        Error18W                        14W                    22%40                           31                        2262                            45                        2780                            55                        31100                          67                        33
I have purchased another kit to experiment with. Did not want to dig this deep or spend this much time, butits gnawing at me.

lu1aax, 2021-07-27:
> Hi.
> The 3.0 firmware uses 1.3db fiber loss by default.
> Renew it to latest, that uses 0 dB by default and a possibility to change the value any time.

Followed your advise. Got lent a PICkit 3, flashed fw 3.2 and fiddled a little with the configuration.

First of all I want to say thank you for sharing such a great work.

I've seen some fuzz about tuning at reduced power settings (QRP). I've seen there is a simple harware mod reducing turns at matching transformer. But this seems to severely reduce the operating measuring range so I tried out other things.

I have a Yaesu FT-817 but I guess some things I will say apply to most "modern" radios. In this radio you can setup 5,3,1, and 1/2 Watt output but if the radio detects a severe SWR mismatch falls back to 1W when set at 3 or 5 W.

There is another problem that could arise when you use not proper diodes. 1N5711 are recommended but reading along this issues reporting board it seems like many units and kits from China don't use these diodes directly affecting the precision of power readings in this device.

In my case SMD diodes came soldered. I don't think they are 1N5711 or similar... they seem to work OK at lower power levels resulting in accurate readings... above 10 watts there is a clear bias with readings below the actual power (-10% at most). I can live with that, but I will try replacing those 2 diodes with 1N5711 or some other diode in spec for the HF+6m band. I am also trusting the MFJ power meter of my manual tuner, I will soon be able to use a Bird wattmeter to confirm all this. This doesn't seem to compromise SWR measuring which is very good, an essential to getting a good match with the tuner.

Now I have found a solution that works for me, letting me tune at QRP levels with my FT-817. It's very simple. I just use a 0 watt level at cell 5.  This is working for me and there is a small side effect: the tuner doesn't wait for any power to be present at input and when set to TUNE mode will start trying to tune on no power. I can live with that too. **I think this side effect can be avoided if  there is an explicit 0W check before starting (if min level = 0 W and actual power = 0W, then don't start trying to tune)**

I have also found fw 3.2 seems to find better matchings than 3.0, but I don't know if the feed line loss now set to 0 could affect this.

I do not comment these things here to be blindly followed. These are results obtained with my particular hardware setup at my station.  I will be trying it out other situations in the next days/weeks.

w0ati, 2021-07-27:
Just completed an interesting set of measurements. I have been trying to explain why my power readingson the ATU-100 are so far off (low).I first applied a DC voltage to the PIC ADC input. This simulates a voltage from thebridge. I swept the voltage such that the ATU display read from 10W to the maximum 165W (code rev 3.2)I then built a bread board of the bridge with the detector diodes. I applied power (14MHz) from 10 to 140Wand read the detector output voltage (what would be applied to the PIC ADC). Plotting the two curves together shows they do not match up. It indicates the power reading will be low.
 I then noticed the following note in the paper work that came with a second ATU-100 kit I purchased. " if PWR not Correct adjust 21 10-> 8 or 12". 21 refers to the Tandem Match bridge transformer. I think this is saying change the turns ratio of the transform for 10 turns to 8 or 12 turns. If I scale the data I collected for a turns ratio of 8 rather than 10, the two curve's would matchup. I have seen a lot of discussion about changing the ratio for QRP operations. But nothing aboutmatching it to the PIC ADC  voltage to power curve.
Alternatively, I could change the resistor divider between the diodes and the ADC.

Dfinitski, 2021-07-28:
Hi.
You can change tandem match ratio in the cell as you want for matching curves with no changing actual ratio .

w0ati, 2021-07-28:
Assuming you have a way to reprogram the PIC.And it is unclear how the scaling (changing the cell) changes the curve shape.

Dfinitski, 2021-07-28:
The tandem match works as a voltage transformer with specified ratio. That means output voltage less that antenna's voltage in 5(10) times, depending to physical ratio.
When you change the ratio value in the cell, you change the multiplication in the calculation formula and shift the curve up or down.

lu1aax, 2021-07-28:
> The tandem match works as a voltage transformer with specified ratio. That means output voltage less than antenna's voltage in 5(10) times, depending to physical ratio.
> When you change the ratio value in the cell, you change the multiplicator in the calculation formula and shift the curve up or down.

Tried this increase in the value of cell 31 to value 11 (actual turns, 10) and now the power reading for 10+ watts fits very well with my (trusted) MFJ wattmeter readings. The side effect (expected) is power readings at low power are a bit inflated (FT-817 output at full power read exactly 5W before, now reads 5.8W which I find absolutely acceptable).

This is a good tip to deal with the nonlinearity of the hardware my setup is suffering.

w0ati, 2021-07-28:
There is the magic numbers 5 and 10 again. Why does 10 keep coming up when 8T on the tandem match withthe default cell values for code Rev 3.0 and 3.2 produces much more accurate power readings?

Dfinitski, 2021-07-28:
The calculation formula has also numerous correction coefficients, Wich I
had by measuring and my setup has very good power meter accuracy. You
probably have to have diodes exactly like main to reach the same result  .

lu1aax, 2021-07-29:
> The calculation formula has also numerous correction coefficients, Wich I had by measuring and my setup has very good power meter accuracy. You probably have to have diodes exactly like main to reach the same result .

Are you using 1N5711?

Dfinitski, 2021-07-29:
No, I prefer bat41

lu1aax, 2021-08-05:
> No, I prefer bat41
> […](#)

Well I got a couple of ST BAT 41 diodes and replaced the SMD diodes that came with the kit from China. I am getting the same power readings at the same frequencies as the original diodes. The SMD diodes I removed are marked L9 which I looked up and found out that it corresponds to a variety of diodes, may be they are SMD BAT41 or similar.

I wonder how could I calibrate my device to match a good reference instrument like a Bird wattmeter. As I described earlier, I am getting correct power readings at low power levels but from 10W up the readings start to be short, like approximately 10% less at 100W.

Dfinitski, 2021-08-06:
HI.
That is very vierd .
You can change the correction coefficients you can find in main.h and looks like this

 ```
int correction(int input) {
     //
     if(input <= 80) return 0;
     if(input <= 171) input += 244;
     else if(input <= 328) input += 254;
     else if(input <= 582) input += 280;
     else if(input <= 820) input += 297;
     else if(input <= 1100) input += 310;
     else if(input <= 2181) input += 430;
     else if(input <= 3322) input += 484;
     else if(input <= 4623) input += 530;
     else if(input <= 5862) input += 648;
     else if(input <= 7146) input += 743;
     else if(input <= 8502) input += 800;
     else if(input <= 10500) input += 840;
     else  input += 860;
     //
     return input;
}
```

First numbers are an input in millivolts, seconds are correction millivolts to needed value for correct indication.

g4tsn, 2021-09-23:
I have an issue where the SWR auto tune is in  a different position to minimum SWR on another external meter and this is confirmed by the SWR the radio sees on its SWR reading being same as external meter  only slight difference but would be nice to find cause or correct this.   
If I go to auto tune note the readings of L and C and then go to manual Test mode and go to the same values I get the same readings but can now alter the L and C maybe a couple of steps either way to get good reading on the power meter and radio 
the tuners bridge does indeed now not read the minimum it found on the auto tune of course 

I have tried changing diodes and have checked the ferrites there is always this difference between the minimum anyone any thoughts on this please ...other wise it works great ......It would be nice to be able to somehow go into manual tune test mode within the operation rather than having to power off tuner .....great project 73, Jon G4TSN

w0ati, 2021-09-23:
I see the same thing. My experience is the ATU swr reading is only useful in finding the minimum. The ATU can not be counted on to provide an accurate swr or power reading. I also find swr changes as the transmitted power changes.

g4tsn, 2021-09-23:
Just trying to understand what could cause this slightly what could cause this seemingly miss-placed dip in SWR compared with other meters .....is this likely to be the windings ?  or the ferrite ? or diodes the power level reading seems to be about right.  

considering changing the diodes again and checking the windings ?  determined to get this right 

Thanks 

Jon G4TSN

Dfinitski, 2021-09-23:
Other gauges may have up to 30% accuracy, ATU 100 with original diodes has
10% and even better.

Dfinitski, 2021-09-23:
> Point taken on accuracy, would have thought the dip would be in same place
> ? it is with radio and other external meter including now Bird 43 ...just
> the ATU bridge seems to have slightly different result on my version
> .......I must have something not quite correct I guess ? just what ?
>
> Is there any way we could add feature access to the manual tune with the
> buttons from main atu without re powering into test mode ? I have the
> compiler and can compile the source but no idea where to add this ....would
> have a go if someone can start me off
>
> Thanks G4TSN

ATU 100 could not reach the best position for some reason. The algorithm
does not check all the possible L and C combination , it stops when the
curve becomes flat enough or changes it's direction.
Unfortunately I did not hear any values from you. What automatic mode
reaches SWR ? What in manual mode ? What is difference SWR measurement ?
What is difference in the power measurement in procents ?
So I can't tell you something really helpful, just common words.

g4tsn, 2021-09-23:
Hi David 

Sorry my description is not good,  I will try to give and examples on 14 Mhz    auto tuner   tunes and gives 
example1:   SWR = 1.18     C = 160p L = 0.0 uH   reading on tuner display    external meters  read SWR = 2.0  if I set ATU to manual tune and set same values C= 160p  L = 0.0uH  same SWR 1.18  now manually  adjust C =160p  L= 0.1uH   SWR = 1.04  is actually now 1.1 on external SWR meter

example2:  up band a bit  auto tune   SWR 1.02     C =160p  L = 0.0 uH   reading on tuner display  external  meters read SWR = 2.0 again manual tune atu same setting same result  then adjust  C = 144p L = 0.0uH  tuner reads SWR = 1.11  external meters SWR = 1.2   

This is very close but I would like both to be similar  2:1  is not brilliant for the radio as it starts to shutdown power above 2:1 .I would just like to get this slightly closer if i watch the tuner operation on the manual meter it seems to hit the low point and just overshoot sometimes when watching almost like it goes past optimum ?   However as you see values look very good on tuner so it has hit minimum according to its own coupler .......if you have any ideas what to try what would cause this difference something is not correct in the built in bridge ?   Hope that gives more information  73, Jon G4TSN

g4tsn, 2021-09-23:
Hi David 

Interesting, I have just tried another AT-100 different unit and it works 100% in line with my external meters readings so indeed 
so the problem is within the other device. somewhere in the coupler at a guess, I can at least with a working version subsitute things to find whats not working correctly 

I will report back more here when I find what the cause is 

Thanks 

73,

Jon G4TSN   
- [ ]

Dfinitski, 2021-09-27:
Hi.
Keep in mind, the algorithm of tuning will stop when SWR has reached 1.2 or
better. So 1.18 is absolutely normal behaviour. That is the result of fast
tuning process, with no all the L C positions brute force.

g4tsn, 2021-09-27:
Hi David

Thanks

Rebuilt the coupler off board and it works perfectly and in line with my 
other meters too so something is weird still investigating that one but 
clearly the firmware is working nicely

One thing I did notice is that when inductance is 0.00 uH somtimes the 
capacitor connected to either end gives different SWR reading .....due to 
additional path involved presumably and sometimes it stops at the higher 
reading on auto  ? any comments could the tuning check both in these 
circumstances perhaps ?

Other wise all is very good

I would like send display data also to my PC to view when working remotely 
to check what the tuner is doing .......I think I can add a software_uart 
using the port pin if I am not using these RC6 and RC7 potentially
could you tell me where I would you define the pins RC6 RC7 in the source 
code ?  havent found this yet

73,

Jon G4TSN

SWR reading is wrong (#61)

https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/61#issuecomment-927892063

g4tsn, 2021-09-27:
RA6 RA7 i think actually

Dfinitski, 2021-09-27:
Hi.
When you push the bypass button ,the C will connected always to right path
to minimize the SWR . When the tuner is powered off, C is connected to
other, not correct path. That what it is, as is .
Try to find I2C controlled Bluetooth transponder with Virtual com port
protocol for connecting to PC.