Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #62: Fall back to bypass or interference?

harmi99, 2021-07-11:
I modified today ATU100 as per IK3SSG https://www.youtube.com/watch?v=dPys_-_wPcQ video for QRP usage, now with 2x5 turns, EEPROM cell 05 to value 01, c06 to v40, c31 to v05. Previously output power up to 10W was insufficient to initate the tuning reliably. Now it seems much better for 160 and 80m, however for 40m and higher frequencies it tunes first to reasonable SWR and after couple seconds some relays suddenly change ON/OFF positions and I am at SWR>3 and on ATU display I see bypass mode with underscore "_" at the end of the first line. With continuous TX it switches back and forth. Enclosing short video to demonstrate this - after the bypass is somehow started, I manually switched bypass off (having all three buttons) and ATU returns to Auto mode indicated by comma "." at the end of the first line. Do you recommend to increase cell 05 to reduce sensitivity or some other experience? Thanks, Peter. 

https://user-images.githubusercontent.com/87270900/125200132-d7820d00-e269-11eb-8fd7-3ba43d36419c.mp4

harmi99, 2021-07-11:
Faster writing than thinking - 100n capacitors for both Auto and ByPass button solved this issue, now it works fine.... Peter.

Dfinitski, 2021-07-11:
Hi, Peter
Sure, very simple and obvious solution.

frankliao84, 2021-07-28:
> 
> 
> Faster writing than thinking - 100n capacitors for both Auto and ByPass buttons solved this issue, now it works fine.... Peter.

would u mind tell us the postition number you changed to 100nF cap?

harmi99, 2021-07-28:
frankliao, there is no position, this is not related to caps connected to relays, it is just about parallel caps to the switches/buttons for Auto and ByPass function as it is described in the ATU-100_Extended_Board_User_Manual_eng.pdf in the section Additional Buttons.  ph

frankliao84, 2021-07-29:
> 
> 
> frankliao, there is no position, this is not related to caps connected to relays, it is just about parallel caps to the switches/buttons for Auto and ByPass function as it is described in the ATU-100_Extended_Board_User_Manual_eng.pdf in the section Additional Buttons. ph

OK.