Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #65: Function request: min pwr/max power for tuning setting from device

lu1aax, 2021-08-03:
Hi, I think it would be great if the values for minimum/maximum power for tuning could be set from the device, in a similar way to setting up feeder loss in dB in firmware 3.2. 

This would let configure the device on the field without needing a PIC programmer to change theses values.

Mauricio LU1AAX

Dfinitski, 2021-08-04:
hi, Mauricio
I don't see any reason to change these values in the field, 
The values by default are fully suitable for 99% users.

lu1aax, 2021-08-05:
> hi, Mauricio
> I don't see any reason to change these values in the field,
> The values by default are fully suitable for 99% users.

Well I intend to use this tuner with a ft-817nd with 5W max out power. Only way to reliably tune without changing default hardware configurations (like turns at matching transformer kepto to 10) is to set min power to 0. I also use it with more poweful radios and would like it to tune with 10/15W of power. Accordlingly it would be nice to be able to set up max power (default no limit) in a similar way.

Dfinitski, 2021-08-06:
Hi, Mauricio
This is a 100W tuner, it has been designed especially for 100 W.
For QRP operations I have designed another tuner  https://github.com/Dfinitski/ATU-10-10W-QRP-antenna-tuner
The best way is to use right tuner for right power.