Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #67: What is the max fwd power level?

frankliao84, 2021-08-10:
I tried to make it work under 300W average with CW mode. Is that possible? If not, what is the limit for that? cap max voltage or the  tandem matching core?

Dfinitski, 2021-08-10:
Hi. This tuner by default can measure up to 150W , bat you can change the
tandem match ratio and some cells in the firmware for work up to 1kw and
more. See the manual, cells description.

pimol, 2021-08-10:
100W

Pietro I2OIM

frankliao84, 2021-08-11:
> Hi. This tuner by default can measure up to 150W , bat you can change the tandem match ratio and some cells in the firmware for work up to 1kw and more. See the manual, cells description.
> […](#)

Thanks. I just worried about the insertion loss of 1000w might burn the core. Other issues like feedback voltage limit could be fixed with minor hw and sw change.

Dfinitski, 2021-08-11:
There are some modifications of this tuner with large inductors core and
capacitors for 1kw operating power. The most important the processor and
the firmware can do that, you are right.  You can find that pcb on eBay,
from Russians sellers.

G0NMY, 2021-08-11:
Are the contacts in those small relays capable of handling 1KW?? They are
quite small arent they???
I would think that better quality relays would be required for such power
levels???

Dfinitski, 2021-08-11:
Better one time to see
https://m.youtube.com/watch?v=fCgariqxoNk&feature=youtu.be