Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #68: Various problems - help!

delight-f, 2021-08-14:
Again, thanks for this project. My problems are almost certainly user error.The MCU is running v.3.2 - bought from aliexpress. I'm a relatively new ham, less than a year. 

1. As I increase power (say from ten Watts all the way to 50w), I notice the swr increasing - is this normal?? If so, why?

2. At some point, even when auto mode is disabled (dot not visible), the tuner will start tuning automatically. I think this led to problem 3.

3. When I was using it last, I tuned at low power and turned auto mode off. I then switched to 100w. The tuner flicked into bypass mode briefly, then tuned as I was sending. Something inside (I think one of the coils) got incredibly hot - to the point where it deformed my PLA printed housing. All my soldering appears OK. I wonder if it was because it went into tuning mode somehow as I was transmitting at that power.

Dfinitski, 2021-08-14:
Hi.
1. It is normal, nonlinear diode characteristics, something become hot.
2.  I don't know, RF interferences on the button wires ?
3. Chinese Amidon cores with losses on RF.

delight-f, 2021-08-14:
Thanks so much for your feedback. I appreciate your quick responses.

The RF on the buttons is a real possibility. They are connected by short flying leads to the pads. Do you have any suggestions on how to reduce the rf interference on these?

Dfinitski, 2021-08-14:
Short pins to gnd with 0.1uf capacitors directly on the pads

delight-f, 2021-08-24:
I've installed the caps to ground and the button activating spontaneously is not a problem. However, I have real issues with RF causing interference with the screen - a typical small OLED. It scrambles the readout so it is not legible. It is mounted in a 3d printed plastic housing. Any suggestions? @Dfinitski

pu7vll, 2023-01-11:
Good evening, Bought with a friend recently two atu 100 on Aliexpress for our USDX, but we both see on the TUNE screen and we do not hear the relé working, ie is not docking! Could someone guide us, we are using version 3.1 by N7DDC. TKS

Dfinitski, 2023-01-11:
https://groups.io/g/ATU100/topics

Dfinitski, 2023-01-11:
Hi. Ask in our group. 
What power and SWR value do you see on the display ?

pu7vll, 2023-01-11:
less than 4w and with 8 SWR test. TUNE appears but the relays do not work

pu7vll, 2023-01-11:
> Hi. Ask in our group. 
> What power and SWR value do you see on the display ? 

less than 4w and with 8 SWR test. TUNE appears but the relays do not work